micropython/micropython #5874: OS X Catalina, mpy-cross: python3 ../py/makeversionhdr.py build/genhdr/mpversion.h

LDS79, 2020-04-06:
Hello, 
I'm not able to built mpy-cross,
result of V=1: python3 ../py/makeversionhdr.py build/genhdr/mpversion.h
SO: macOS 10.15.4
I have already read:  #4281 and #4455 they don't seem to help me and have been posted more then year ago

what could I try? Thanksssss

dlech, 2020-04-06:
What is the error message?

LDS79, 2020-04-06:
This is the error:
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.

From what I understand it is very common issues but I don’t understand how to resolve it

dlech, 2020-04-06:
That is not an error message, that is just informational.

LDS79, 2020-04-06:
I receve this: python3 ../py/makeversionhdr.py build/genhdr/mpversion.h
Nothing more
How can I make more investigation?

dlech, 2020-04-06:
Make sure `python3` is installed and in your `PATH`.

dlech, 2020-04-06:
Also try running `make -C mpy-cross clean`

davehylands, 2020-04-07:
I didn't have any issues building mpy-cross under Catalina 10.15.4, but I ran into an issue building the main executable.

Make sure you have pkg-config installed: `brew install pkg-config`
Make sure you have libffi installed: `brew install libffi`

I then ran `pkg-config --libs libffi` and got the error `No package 'libffi' found`.

I had to add a special configuration for pkg-config.
```
export PKG_CONFIG_PATH="$(brew --prefix libffi)/lib/pkgconfig/"
```
and then rerunning `pkg-config --libs libffi` reported: `--L/usr/local/Cellar/libffi/3.2.1/lib -lffi`

After that I was able to build ports/unix (Note: I didn't need to do any of the above to build mpy-cross).

stinos, 2020-04-07:
^ this. When you cannot build something you can go look in travis.yml to see how it gets done from scratch. IIRC there's a PR which is about that, e.g. having those instructions in a Makefile so they're easily usable.

LDS79, 2020-04-08:
@dlech  if I perform: "make -C mpy-cross clean"  and "make mpy-cross", the result I alway the same: 
(build-venv) lucaxxxxx@iMac-di-Luca mpy-cross % make mpy-cross         
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
mkdir -p build/genhdr
GEN build/genhdr/mpversion.h
GEN build/genhdr/moduledefs.h
GEN build/genhdr/qstr.i.last
etc etc etc ....

if I repeat the command: make V=1 mpy-cross the result:
python3 ../py/makeversionhdr.py build/genhdr/mpversion.h

dlech, 2020-04-08:
It sounds like it completed successfully the first time and there was nothing to do the second time.

davehylands, 2020-04-08:
I can confirm that I see the same thing (which is expected). The second time, it detects that everything is upto date and nothing needs to be rebuilt.

When you run with `V=1` it causes all of the commands that make runs to be output. Just because you see something doesn't mean that there was an error.

If you just run `make` and only see this output:
```
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
```
that also means that everything is fine (that message is printed everytime make is run, regardless of whether things were successful or not). 

You should be able to run `mpy-cross -h` from within the mpy-cross directory and get output like the following:
```
$ ./mpy-cross -h
usage: ./mpy-cross [<opts>] [-X <implopt>] <input filename>
Options:
--version : show version information
-o : output file for compiled bytecode (defaults to input with .mpy extension)
-s : source filename to embed in the compiled bytecode (defaults to input file)
-v : verbose (trace various operations); can be multiple
-O[N] : apply bytecode optimizations of level N

Target specific options:
-msmall-int-bits=number : set the maximum bits used to encode a small-int
-mno-unicode : don't support unicode in compiled strings
-mcache-lookup-bc : cache map lookups in the bytecode
-march=<arch> : set architecture for native emitter; x86, x64, armv6, armv7m, armv7em, armv7emsp, armv7emdp, xtensa, xtensawin

Implementation specific options:
  emit={bytecode,native,viper} -- set the default code emitter
  heapsize=<n> -- set the heap size for the GC (default 2097152)
```
which confirms that it was built properly.

LDS79, 2020-04-09:
Hi, 
ok but if I perform: make erese 
I get:                                     
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Building with ESP IDF v3
make: *** No rule to make target `erese'.  Stop.

and on my board is still the old firmware.

davehylands, 2020-04-09:
`erese` isn't a valid target. Perhaps you meant `erase`

LDS79, 2020-04-10:
Thanks so much at all.. I really appreciate your help
after my final writing mistake, we can close the ticket! 

Happy Easter to all

closed 2025-10-03 by jonnor