micropython/micropython #5878: esp8266: latest build boot failure, flashbdev.py OSError EIO

irmen, 2020-04-07:
Latest build gives a boot failure on my esp8266:
```MPY: soft reboot
Traceback (most recent call last):
  File "_boot.py", line 13, in <module>
  File "inisetup.py", line 45, in setup
  File "inisetup.py", line 16, in check_bootsec
  File "flashbdev.py", line 13, in readblocks
OSError: [Errno 5] EIO
MicroPython v1.12-363-g05cd61286 on 2020-04-07; ESP module with ESP8266
Type "help()" for more information.
>>> 
```

rolling back to the version from march 28 solves the issue.
Could this perhaps be related to the new filesystem flash format introduced in e0905e85a7ad2961aa9192f6130565860e531ad3 ?   Just a very wild guess

dpgeorge, 2020-04-08:
From the error it looks like your esp8266 module has 1M of flash?  If so you'll need the `GENERIC_1M` build.

dpgeorge, 2020-04-08:
Daily builds for 1M modules are now available at https://micropython.org/download .  Also, instructions there are updated to mention the latest firmware has change the filesystem layout.

irmen, 2020-04-08:
Oh, ehmm I'm pretty much a beginner still.  However, I am building my own port (because I am preprocessing various python modules as mpy's into it) instead of using a prebuilt one.  Flashing and running a build made upto the mentioned commit works fine. I didn't have to select / change anything to use the "GENERIC_1M" build as far as I know.   Was this changed recently ? What should I do to build this specific build, other than just 'make deploy' in the ports/esp8266 directory which I have been doing before?     
I'll investigate a bit more later today myself as well.

irmen, 2020-04-08:
Well, found out that indeed I had to change a tiny thing locally, I guess to select the 1Meg build (which appeared to be the default earlier).
 I've now built the firmware with ``make BOARD=GENERIC_1M deploy`` and reflashed the module.  It now works again!  Thanks for pointing me in the right direction.

closed 2020-04-08 by irmen

dpgeorge, 2020-04-09:
> Well, found out that indeed I had to change a tiny thing locally, I guess to select the 1Meg build (which appeared to be the default earlier)

Yes this also recently changed, the same time the different filesystem + flash layout changed.  Previous default firmware was for modules with 1M or more flash.  New default is for modules with 2M or more flash, and (as you found out) you must now explicitly select `GENERIC_1M` if you have a 1M module.

But, glad it's all working now!

besi, 2024-03-13:
@dpgeorge I did run into an issue that looks related to this thread. If you have time and energy then maybe you might have a look at it. I did not want to add an issue since I'm not sure that it's a bug but maybe rather a misconfiguration on my side.

https://github.com/orgs/micropython/discussions/14077