micropython/micropython #5879: NRF51822 (PCA10028) - no serial console after flashing

scttnlsn, 2020-04-07:
Building the `pca10028` target:

```
cd ports/nrf
make BOARD=pca10028
```

Flashing it with OpenOCD:

```
openocd -f interface/stlink-v2.cfg -f target/nrf51.cfg -c "program ./build-pca10028/firmware.bin verify reset exit"
```

And trying to connect to the serial port:

```
screen /dev/ttyUSB0 115200
```

Serial console is blank though and resetting the board does not do anything.  I've verified that the `MICROPY_HW_UART1_RX` and `MICROPY_HW_UART1_TX` pins in `boards/pca10028/mpconfigboard.h` are correct and I've tried disabling `MICROPY_HW_UART1_HWFC` as well.  No dice.

Any ideas?

jonnor, 2025-10-03:
Is this still a problem on recent MicroPython?

dpgeorge, 2025-10-04:
A lot of changes and fixes have gone into the nrf port since this issue was opened, so it may be fixed.

I'll close this for now.  Feel free to reopen if the issue still needs fixing.

closed 2025-10-04 by dpgeorge