micropython/micropython #5882: extmod/uasyncio gather return_exceptions=True does not return exceptions

peterhinch, 2020-04-08:
This script works as expected under CPython but not under MicroPython:
```python
try:
    import uasyncio as asyncio
except ImportError:
    import asyncio

async def barking(n):
    print('Start barking')
    for _ in range(6):
        await asyncio.sleep(1)
    print('Done barking.')
    return 2 * n

async def foo(n):
    print('Start timeout coro foo()')
    while True:
        await asyncio.sleep(1)
        n += 1
    return n

async def bar(n):
    print('Start cancellable bar()')
    while True:
        await asyncio.sleep(1)
        n += 1
    return n

async def do_cancel(task):
    await asyncio.sleep(5)
    print('About to cancel bar')
    task.cancel()

async def main():
    tasks = [asyncio.create_task(bar(70))]
    tasks.append(barking(21))
    tasks.append(asyncio.wait_for(foo(10), 7))
    asyncio.create_task(do_cancel(tasks[0]))
    res = None
    try:
        res = await asyncio.gather(*tasks, return_exceptions=True)
    except asyncio.TimeoutError:
        print('Timeout')
    except asyncio.CancelledError:
        print('Cancelled')
    print('Result: ', res)

asyncio.run(main())
```
Under CPython:
```
Start cancellable bar()
Start barking
Start timeout coro foo()
About to cancel bar
Done barking.
Result:  [CancelledError(), 42, TimeoutError()]
```
MicroPython:
```
Start cancellable bar()
Start barking
Start timeout coro foo()
About to cancel bar
Cancelled
Result:  None
```
The cancelled exception has propagated to the caller and the timeout exception has apparently vanished.

dpgeorge, 2020-04-09:
Thanks for the report, I can confirm the problem.  Should be fixed by #5899 (gather simply needed to catch CancelledError explicitly, because it's derived from BaseException and not Exception).

peterhinch, 2020-04-10:
That works :+1:

closed 2020-04-10 by peterhinch

dpgeorge, 2021-11-17:
Fixed by 43d08688c398184013bbfb0190e2529e4032e003