micropython/micropython #5884: MicroPython 1.12 source tar ball contains mpconfig.h for 1.11

george-hawkins, 2020-04-08:
I installed MicroPython 1.12 for UNIX using [`pyenv`](https://github.com/pyenv/pyenv) like so:

```bash
$ pyenv install micropython-1.12
Downloading micropython-1.12.tar.xz...
-> https://micropython.org/resources/source/micropython-1.12.tar.xz
Installing micropython-1.12...
Installed micropython-1.12 to /home/ghawkins/.pyenv/versions/micropython-1.12
$ pyenv shell micropython-1.12
$ micropython 
MicroPython v1.11 on 2020-04-08; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> 
```

As you can see it downloads the `micropython-1.12.tar.xz` tarball but when I start it says its version is `v1.11`.

I double checked and downloaded the [1.12 tarball](http://micropython.org/resources/source/micropython-1.12.tar.xz) directly from the [Downloads page](https://micropython.org/download). On untaring it and looking at `micropython-1.12/py/mpconfig.h`, I see:

```C
// Current version of MicroPython
#define MICROPY_VERSION_MAJOR 1
#define MICROPY_VERSION_MINOR 11
#define MICROPY_VERSION_MICRO 0
```

So the tarball seems to have been made at some point in time before commit 1f37194 bumped the minor version to 12.

george-hawkins, 2020-04-08:
I notice that the same is true for the tarball included as part of the [v1.12](https://github.com/micropython/micropython/releases/tag/v1.12) tagged release on GitHub. The contents of the two tarballs are identical (though the one on the Downloads page of the MicroPython site is a `.xz` file while the one GitHub is a `.gz` file).

This is all a bit odd as the v1.12 tag tags the commit I mentioned previously, i.e. 1f37194, that bumps the version to 1.12. So why isn't this seen in the tarballs? Is there something up with whatever process produced them?

dpgeorge, 2020-04-13:
Thanks for the report.  The release process is manual and I obviously made a mistake with v1.12.  The only thing missing was the commit that bumped the version, everything else in the release is correct.

I've now updated all archive files in the download section and on GitHub releases to include the correct version files, and also added a note in the release process to check this in the future.

closed 2020-04-13 by dpgeorge

george-hawkins, 2020-04-13:
Thanks @dpgeorge. Sorry to put you to so much trouble for such a trivial issue but it did just look odd when one installed MicroPython 1.12 via `pyenv` and it then reported its version as 1.11.

I've logged a PR with `pyenv` to update the SHA2 values they currently have associated with these tarballs: pyenv/pyenv#1588.

dpgeorge, 2020-04-13:
> but it did just look odd when one installed MicroPython 1.12 via `pyenv` and it then reported its version as 1.11.

Indeed that's very confusing!