micropython/micropython #5889: IPV6 socket connection

sasilva1998, 2020-04-09:
Hi, I am trying to establish a IPv6 connection with an ESP32 and MicroPython version 1.12. This is the code I am trying:
```
import usocket as socket
import network
wlan=network.WLAN(network.STA_IF)
wlan.active(True)
wlan.connect('ssid', 'pass')
s = socket.socket(socket.AF_INET6, socket.SOCK_STREAM)
s.connect(('2a04:4e42:3d::223',80,0,0))
```
According to the docs, the IPv6 address should have 4 parameters, but I am getting the following error:
```
Traceback (most recent call last):
  File "<stdin>", line 7, in <module>
ValueError: requested length 2 but object has length 4
```
Is it a legit bug or am I doing something wrong? Thanks in advance.

HeMan, 2020-06-03:
Micropython does currently not support IPv6, see #3683

sasilva1998, 2020-06-04:
Thank you for heads up! @HeMan

closed 2020-06-04 by sasilva1998