micropython/micropython #5895: lfs2 mkfs broken

pumelo, 2020-04-09:
`mkfs` is broken for lfs2:

```
MicroPython v1.12-262-g19ea30bdd on 2020-03-19; ESP32 module (spiram) with ESP32
Type "help()" for more information.
>>> import os
>>> os.uname()
(sysname='esp32', nodename='esp32', release='1.12.0', version='v1.12-262-g19ea30bdd on 2020-03-19', machine='ESP32 module (spiram) with ESP32')
>>> import machine
>>> sd = machine.SDCard()
>>> os.VfsLfs2.mkfs(sd)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: function takes 3 positional arguments but 4 were given
```

The same happens with the `RAMBlockDev` example provided here:
http://docs.micropython.org/en/latest/reference/filesystem.html?highlight=flashbdev#custom-block-devices

pumelo, 2020-04-09:
Oh, sorry, my fault.
I used the wrong implementation of the `RAMBlockDev`. If it has the `extended interface` the example works as expected.

Still it seems unfortunate, that littlefs can not be used with an sd-card as the sd card does not support the `extended interface`. 

Edited the original issue description accordingly.

dpgeorge, 2020-04-09:
> Is there a technical reason for not allowing to use littlefs on a sd-card?

littlefs is not designed for large devices, in particular it can get quite slow when a large device starts to fill up.  Also, because SD cards are intended to be also used on a PC, it makes more sense to just use FAT on them.

But, the SD card driver could be made to support the extended interface.  Or, probably better, an adaptor interface could be made which translates extended to standard.