micropython/micropython #5897: right-oriented magic methods are not triggered with non-standard type

apollo2mond, 2020-04-09:
seems related to #2328 but no real fix/build around available, right?
Hi,
I use mpy latest prebuild on esp8266 and sfp (Simple Python Fixed-Point Module (SPFPM))
sfp works as expected on win/linux with python 3.x

BUT sfp has an issue with mpy, e.g. if I multiply a variable from type FXnum with int/float
than it only works (FXnum) * (int/float) but not (int/float) * (FXnum).
As highlighted, it seems that __rmul__ as well as any other right-oriented magic method doesn't work with a non-standard type.

dpgeorge, 2020-04-09:
>  if I multiply a variable from type FXnum with int/float
> than it only works (FXnum) * (int/float) but not (int/float) * (FXnum).

Can you please provide minimal code to reproduce this issue?

apollo2mond, 2020-04-10:
library from here https://github.com/rwpenney/spfpm

from FixedPoint import FXfamily, FXnum

num = FXnum(42)
result = 13 * num # not ok
result = num * 13 # ok

dpgeorge, 2020-04-11:
The above example runs correctly under the latest unix version of MicroPython.  What port/version are you running?

apollo2mond, 2020-04-11:
port: esp8266
version: latest prebuild/dirty build

dpgeorge, 2020-04-12:
> port: esp8266

Ok, so this port does not have reverse special methods enabled.  If you are able to build the firmware yourself then add `#define MICROPY_PY_REVERSE_SPECIAL_METHODS (1)` to `ports/esp8266/mpconfigport.h` to enable it.

peterhinch, 2020-04-13:
This relates to https://github.com/micropython/micropython/issues/5564. Python code portability is becoming problematic.

apollo2mond, 2020-04-14:
I dislike the approach of any hidden config flags and the current related config handling makes less sense for me! 
I propose that ALL hidden config options will be visible in mpconfigport.h and reasons to switch these flags on/off are mentioned/highlighted.

tve, 2020-04-14:
@apollo2mond what do you mean by "the current related config handling"?

Can you provide examples of "hidden config options" that are not "visible in mpconfigport.h"? I know that a small number are in the makefile for technical reasons but I suspect you mean something else? Also, the one mentioned here is in mpconfigport.h: " #define MICROPY_PY_REVERSE_SPECIAL_METHODS (1) to ports/esp8266/mpconfigport.h t".

I'm simply trying to understand what you mean... I agree about "reasons to switch these flags on/off are mentioned/highlighted"

apollo2mond, 2020-04-15:
e.g. what about the option for float64?
AFAK it is for esp32 possible and for esp8266 not, BUT this status is invisible in mpconfigport for both cases!?

tve, 2020-04-15:
> Sure, e.g. what about the option for float64?
> AFAK it is for esp32 possible and for esp8266 not, BUT this status is invisible in mpconfigport for both cases!?
> .. and the mentioned MICROPY_PY_REVERSE_SPECIAL_METHODS flag must be added at least for esp8622 and esp32?

I haven't use float on the esp8266, but I do see:
https://github.com/micropython/micropython/blob/master/ports/esp8266/mpconfigport.h#L92
and
https://github.com/micropython/micropython/blob/master/ports/esp32/mpconfigport.h#L47

I don't see the SPECIAL_METHODS in the esp8266 mpconfigport.h... feel like creating a PR that adds the missing defines? I believe one of the issues is that boards can override the default so you end up with a lot of `#ifndef - #define - #endif` in mpconfigport.h, which ends up making an unwieldy list. MIght be better to have a script that updates a master list in the docs?

jimmo, 2020-04-15:
Counterargument: all the flags are defined in mpconfig.h, they have comments there explaining what they do and what their valid values are.

Why duplicate all of the entries in each of the port mpconfigport.h, when you have to look at mpconfig.h anyway for documentation? And if I add a new config flag for a new feature, I don't want to have to update every port.

I'd suggest instead the opposite -- to _remove_ any entries in mpconfigport.h that match the default (i.e. get the port and board configs down to the absolute minimum). And perhaps add a comment to the top of every mpconfig*.h to say "Look in mpconfig.h for details".

dpgeorge, 2020-04-16:
To address the original issue here I suggest we enable `MICROPY_PY_REVERSE_SPECIAL_METHODS` on the "big" bare-metal ports (stm32, esp8266, esp32), which should be ok for esp8266 because it now has more executable flash space (just 2M+ modules).

OlivierLenoir, 2021-02-05:
[Pure MicroPython Matrix (mathematics)](https://forum.micropython.org/viewtopic.php?f=15&t=9613#p54375)
I'm getting this error _TypeError: unsupported types for __add__: 'int', 'Matrix'_. It seem to be link to def ___radd__(self, other):_. In fact my class Matrix works well with python 3.7.3 but not with micropython v1.13 or v1.14.
Full project is available [here](https://gitlab.com/olivier.len02/MicroPython-Matrix).

I'm using ESP32 board without SPIRAM and I plan to use the recent Raspberry Pi PICO.
At that stage I don't feel confident to custom build my own version of MicroPython.
It would be a good idea to enable, __radd__(), __rsub__(), __rmul__()...

dpgeorge, 2022-07-08:
Reverse special methods were enabled on esp8266 GENERIC board in 9f8087b448890417d5aa86551957a2100616bb8e

closed 2022-07-08 by dpgeorge