micropython/micropython #5898: extmod/uasyncio Tracebacks give incorrect line numbers

peterhinch, 2020-04-09:
The following traceback occurred (due to a bug in my stream.write()):
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "iorw.py", line 98, in <module>
  File "uasyncio/core.py", line 1, in run
  File "uasyncio/core.py", line 1, in run_until_complete
  File "uasyncio/core.py", line 1, in run_until_complete
  File "iorw.py", line 93, in sender
  File "uasyncio/stream.py", line 1, in stream_awrite
  File "uasyncio/stream.py", line 1, in drain
TypeError: function takes 4 positional arguments but 2 were given
```

dpgeorge, 2020-04-09:
Do you have uasyncio frozen in the firmware?  Because it uses bytecode optimisation and does not emit line number info, when frozen.

peterhinch, 2020-04-10:
Yes. Thanks for the explanation. The kind of bug that causes `uasyncio` to barf is uncommon.

But is this now true for all frozen bytecode? If so, it will make development on ESP8266 difficult - can the behaviour be controlled at build time?

tve, 2020-04-10:
IIRC it's determined by the optimization level passed to mpy-cross

peterhinch, 2020-04-10:
Ah, got it. `extmod/uasyncio/mainfest.py` has `opt=3`. Thanks for the explanation.

closed 2020-04-10 by peterhinch