micropython/micropython #5900: UDP server for new uasyncio code

perbu, 2020-04-09:
Hello.

I saw there was a new uasyncio library the other day. I also saw there was no UDP support in it so I wrote my own library. It is quite simplistic, it basically allows me to register a callback which gets the result from recv(). If the callback returns anything it is sent back as a reply the UDP packet. I rely on uselect.poll() to see if there are packets available.

The code good enough for me and my silly projects but if you want me to I'd be happy to whip it into shape and send a PR. Let me know if you want me to. The current code is here. https://github.com/perbu/dgram

Thanks for a wonderful tool.
Per

perbu, 2021-07-05:
According to @oclyke I should open this again. It seems this is still a missing feature in Micropython.  See https://github.com/perbu/dgram/issues/1 for a bit more context.

dpgeorge, 2021-07-05:
Would be great to add support for this to the core code.  But the API really needs to match CPython.

IhorNehrutsa, 2023-12-11:
docs/esp32: UDP MULTICAST client/server example. #13166

jonnor, 2024-09-29:
Related is the low-level support for a callback on UDP receive, https://github.com/micropython/micropython/issues/3594