micropython/micropython #5901: Update to 3.3.2 SDK

petrkr, 2020-04-10:
Espressif released few days ago tag v3.3.2. I tried simple compile it, but there seems to be lot of changes related to BLE.

Consider update to 3.3.2 to be more closer to upstream SDK. Also it includes lot of fix for binary wifi blobs. Maybe it can fix #5554 problems too...

https://github.com/espressif/esp-idf/tree/v3.3.2

tve, 2020-04-10:
Q: why spend more effort on 3.3 as opposed to fixing anything that might not work with 4.0 or working on 4.1? (I'm not using 3.3 at all, so I don't know what I'm missing...)

petrkr, 2020-04-10:
Because on 4.x does not work wifi and ble together? Or that is some different problem?

dpgeorge, 2020-04-11:
It's not hard to do, see #5904

petrkr, 2020-04-14:
I tested as follows:
SDK 3.3.1 with hard-coded messy "bugfix" by https://github.com/petrkr/micropython/commit/b3ee70dd50143660a9e3b9c1d06f946d7376ab2e and bcn timeout dissapeared
SDK 3.3.2 works fine without any other code change
SDK 4.0 act same as 3.3.1 and produces bcn_timeout errors

For me it's better to do this update to 3.3.2 than wait with bcn errors until 4.x will be stable

dpgeorge, 2020-04-14:
> SDK 3.3.2 works fine without any other code change

Ok, great, thanks for testing.  Then I think it's worth updating to IDF v3.3.2.

dpgeorge, 2020-04-14:
Done in f534b9976511013bc4350826ea49ba8b6a92ce89

closed 2020-04-14 by dpgeorge