micropython/micropython #5907: esp32: viper tests fail after a while

tve, 2020-04-11:
If I execute a micropython/viper test on an esp32 in a loop it starts failing after a while:
```
> while true; ./run-tests --device /dev/ttyUSB0 --target esp32 micropython/
viper_ptr32_load.py; end
pass  micropython/viper_ptr32_load.py
1 tests performed (4 individual testcases)
1 tests passed
[...]
pass  micropython/viper_ptr32_load.py
1 tests performed (4 individual testcases)
1 tests passed
FAIL  micropython/viper_ptr32_load.py
1 tests performed (4 individual testcases)
0 tests passed
1 tests failed: viper_ptr32_load
skip  micropython/viper_ptr32_load.py
0 tests performed (0 individual testcases)
0 tests passed
1 tests skipped: viper_ptr32_load
```
It takes less than a minute to get there. I noticed because running the full test suite shows this, e.g.:
```
> ./run-tests --device /dev/ttyUSB0 --target esp32 -d micropython
[...]
pass  micropython/viper_misc_intbig.py
pass  micropython/viper_ptr16_load.py
pass  micropython/viper_ptr16_store.py
pass  micropython/viper_ptr32_load.py
pass  micropython/viper_ptr32_store.py
FAIL  micropython/viper_ptr8_load.py
FAIL  micropython/viper_ptr8_store.py
FAIL  micropython/viper_subscr.py
FAIL  micropython/viper_try.py
FAIL  micropython/viper_types.py
FAIL  micropython/viper_with.py
65 tests performed (411 individual testcases)
59 tests passed
4 tests skipped: heap_locked heapalloc_bytesio2 meminfo memstats
6 tests failed: viper_ptr8_load viper_ptr8_store viper_subscr viper_try viper_types viper_with
```
The tests that fail at the end are because they're at the end, not because they fail per-se.

tve, 2020-04-11:
The following tests show the same symptoms:
```
basics_int_big_mul.py
basics_int_big_div.py
```
running just a random test in a loop doesn't seem to bring up the problem.
I did test with both esp-idf v3 and v4, they perform the same.

dpgeorge, 2020-04-13:
> If I execute a micropython/viper test on an esp32 in a loop it starts failing after a while

This is probably due to out-of-memory: the native emitter allocates IRAM to store the generated code, and I don't think it's ever freed.

tve, 2020-04-13:
> > If I execute a micropython/viper test on an esp32 in a loop it starts failing after a while
> 
> This is probably due to out-of-memory: the native emitter allocates IRAM to store the generated code, and I don't think it's ever freed.

Do you consider this a bug or a feature ;-)

It's probably a non-issue in normal use, but it makes it impossible to run the test suite as-is. Thoughts?

dpgeorge, 2020-04-13:
An easy way to fix it is free all allocated IRAM on soft reset (so a list will need to be kept of all allocations made).  This will probably fix most use cases (including running the test suite).

A harder (and arguably more correct) way is to have a finaliser for native code run on garbage collection which frees the IRAM.