micropython/micropython #5909: embedding example is broken

hellozyemlya, 2020-04-11:
Can't build embeddine example with following output:
```
workstation@workstation:~/dev/lego/micropython$ make -C examples/embedding
make: Entering directory '/home/workstation/dev/lego/micropython/examples/embedding'
cc -std=c99 -I. -I../.. -DNO_QSTR   -c -o hello-embed.o hello-embed.c
In file included from ../../py/reader.h:29:0,
                 from ../../py/lexer.h:33,
                 from ../../py/compile.h:29,
                 from hello-embed.c:31:
../../py/obj.h:733:66: error: unknown type name ‘mp_rom_error_text_t’; did you mean ‘mp_rom_map_elem_t’?
 mp_obj_t mp_obj_new_exception_msg(const mp_obj_type_t *exc_type, mp_rom_error_text_t msg);
                                                                  ^~~~~~~~~~~~~~~~~~~
                                                                  mp_rom_map_elem_t
../../py/obj.h:734:71: error: unknown type name ‘mp_rom_error_text_t’; did you mean ‘mp_rom_map_elem_t’?
 mp_obj_t mp_obj_new_exception_msg_varg(const mp_obj_type_t *exc_type, mp_rom_error_text_t fmt, ...); // counts args by number of % symbols in fmt, excluding %%; can only handle void* sizes (ie no float/double!)
                                                                       ^~~~~~~~~~~~~~~~~~~
                                                                       mp_rom_map_elem_t
In file included from ../../py/bc.h:30:0,
                 from ../../py/emitglue.h:30,
                 from ../../py/compile.h:31,
                 from hello-embed.c:31:
../../py/runtime.h:165:59: error: unknown type name ‘mp_rom_error_text_t’; did you mean ‘mp_rom_obj_tuple_t’?
 NORETURN void mp_raise_msg(const mp_obj_type_t *exc_type, mp_rom_error_text_t msg);
                                                           ^~~~~~~~~~~~~~~~~~~
                                                           mp_rom_obj_tuple_t
../../py/runtime.h:166:64: error: unknown type name ‘mp_rom_error_text_t’; did you mean ‘mp_rom_obj_tuple_t’?
 NORETURN void mp_raise_msg_varg(const mp_obj_type_t *exc_type, mp_rom_error_text_t fmt, ...);
                                                                ^~~~~~~~~~~~~~~~~~~
                                                                mp_rom_obj_tuple_t
../../py/runtime.h:167:35: error: unknown type name ‘mp_rom_error_text_t’; did you mean ‘mp_rom_obj_tuple_t’?
 NORETURN void mp_raise_ValueError(mp_rom_error_text_t msg);
                                   ^~~~~~~~~~~~~~~~~~~
                                   mp_rom_obj_tuple_t
../../py/runtime.h:168:34: error: unknown type name ‘mp_rom_error_text_t’; did you mean ‘mp_rom_obj_tuple_t’?
 NORETURN void mp_raise_TypeError(mp_rom_error_text_t msg);
                                  ^~~~~~~~~~~~~~~~~~~
                                  mp_rom_obj_tuple_t
../../py/runtime.h:169:44: error: unknown type name ‘mp_rom_error_text_t’; did you mean ‘mp_rom_obj_tuple_t’?
 NORETURN void mp_raise_NotImplementedError(mp_rom_error_text_t msg);
                                            ^~~~~~~~~~~~~~~~~~~
                                            mp_rom_obj_tuple_t
<builtin>: recipe for target 'hello-embed.o' failed
make: *** [hello-embed.o] Error 1
make: Leaving directory '/home/workstation/dev/lego/micropython/examples/embedding'
```

hellozyemlya, 2020-04-11:
After applying following diff
```
diff --git a/examples/embedding/hello-embed.c b/examples/embedding/hello-embed.c
index 2000b703c..255e2c056 100644
--- a/examples/embedding/hello-embed.c
+++ b/examples/embedding/hello-embed.c
@@ -28,6 +28,9 @@
 #include <stdio.h>
 #include <stdlib.h>
 
+#define NORETURN
+#include "mpconfigport.h"
+#include "py/misc.h"
 #include "py/compile.h"
 #include "py/runtime.h"
 #include "py/gc.h"
diff --git a/examples/embedding/mpconfigport_minimal.h b/examples/embedding/mpconfigport_minimal.h
index fa52be4ad..9038849e7 100644
--- a/examples/embedding/mpconfigport_minimal.h
+++ b/examples/embedding/mpconfigport_minimal.h
@@ -27,6 +27,7 @@
 // options to control how MicroPython is built
 
 #define MICROPY_ALLOC_PATH_MAX      (PATH_MAX)
+#define MICROPY_ENABLE_DYNRUNTIME   (1)
 #define MICROPY_ENABLE_GC           (1)
 #define MICROPY_ENABLE_FINALISER    (0)
 #define MICROPY_STACK_CHECK         (0)
```
to resolve mentioned above issues, it fails with 
```
workstation@workstation:~/dev/lego/micropython$ make -C examples/embedding
make: Entering directory '/home/workstation/dev/lego/micropython/examples/embedding'
make -f ../../examples/embedding/Makefile.upylib MPTOP=../..
make[1]: Entering directory '/home/workstation/dev/lego/micropython/examples/embedding'
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
make[1]: *** No rule to make target 'ports/unix/file.c', needed by 'build/genhdr/qstr.i.last'.  Stop.
make[1]: Leaving directory '/home/workstation/dev/lego/micropython/examples/embedding'
Makefile:8: recipe for target '-lmicropython' failed
make: *** [-lmicropython] Error 2
make: Leaving directory '/home/workstation/dev/lego/micropython/examples/embedding'
```
Unfortunately I am not an expert in makefile black magic and can't patch it on my own :(

dlech, 2020-04-11:
`ports/unix/file.c` was replaced recently with ` extmod/vfs_posix_file.c`

See https://github.com/micropython/micropython/pull/5769

tve, 2020-04-11:
Just looking from the distance.... it seems this is related to the recently introduced compressed error strings. Maybe you can try to `git checkout v1.12` and rebuild.

hellozyemlya, 2020-04-12:
@dlech @tve thanks a lot for suggestions, created PR https://github.com/micropython/micropython/pull/5912 to fix this issue.

dpgeorge, 2021-04-09:
Should be fixed by 42035e5eded531d41319ffde999fa5a97478c872

closed 2021-04-09 by dpgeorge