micropython/micropython #5910: ESP32 SPI bits=0 results in IntegerDivideByZero

morganrallen, 2020-04-11:
Simple typo led to this discover...

```
hspi = SPI(2, 10000000, sck=SCK, mosi=MOS, phase=0, bits=0, firstbit=0)
```

Full dump
```
Guru Meditation Error: Core  1 panic'ed (IntegerDivideByZero). Exception was unhandled.
Core 1 register dump:
PC      : 0x400f974f  PS      : 0x00060530  A0      : 0x801b6510  A1      : 0x3ffd0130  
A2      : 0x3ffc20f0  A3      : 0x00000088  A4      : 0x3ffe5870  A5      : 0x00000000  
A6      : 0x00000440  A7      : 0xfffffffe  A8      : 0x00000000  A9      : 0x3ffd0120  
A10     : 0x3ffd0130  A11     : 0x00000000  A12     : 0x00000028  A13     : 0x3ffd0158  
A14     : 0x0000002a  A15     : 0x00000002  SAR     : 0x00000006  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

Backtrace: 0x400f974c:0x3ffd0130 0x401b650d:0x3ffd0180 0x400f1803:0x3ffd01a0 0x400e3751:0x3ffd01d0 0x400df7d9:0x3ffd01f0 0x400df8fe:0x3ffd0210 0x400ed281:0x3ffd0230 0x400e38a8:0x3ffd02d0 0x400df7d9:0x3ffd0300 0x400df802:0x3ffd0320 0x400df86f:0x3ffd0340 0x400ea4ed:0x3ffd03d0 0x400ea82a:0x3ffd0400 0x400df97e:0x3ffd04e0 0x400ed51c:0x3ffd0520 0x400e38a8:0x3ffd05c0 0x400df7d9:0x3ffd0620 0x400df802:0x3ffd0640 0x40103b84:0x3ffd0660 0x40103dad:0x3ffd06f0 0x40103e0b:0x3ffd0710 0x400f5c8a:0x3ffd0730 0x4008fc19:0x3ffd0760

Rebooting...
```

Build against 740946736

jonnor, 2024-10-01:
Hi thank you for reporting. Does this still happen in recent MicroPython, such as 1.23 or later?

agatti, 2024-10-01:
Still happens with current `master`.  I'll send a PR for this.

PR #15950 (2024-10-30): esp32/machine_hw_spi: Reject invalid number of bits in constructor.
### Summary

This commit adds an extra bit of parameters validation to the SPI bus constructor on ESP32.  Passing 0 as the number of bits would trigger a division by zero error when performing read/write operations on an SPI bus created in such a fashion.

This fixes #5910.

### Testing

This was tested locally on an ESP32-D0WD:

```
MicroPython v1.24.0-preview.363.g17d823445.dirty on 2024-10-01; Generic ESP32 module with ESP32
Type "help()" for more information.
>>> import machine
>>> machine.SPI(2, 10000000, sck=10, mosi=20, [... cut]
closed 2024-10-30 by dpgeorge