micropython/micropython #5911: extmod/uasyncio: Exception handling differences to CPython

kevinkk525, 2020-04-12:
During a message exchange with @peterhinch I (re)discovered some incompatibilities I'd like to share. At first it looked like some easy and small difference but the more I dug into it, the more suprised I got.. (keep reading :D )
Even though they might not get "fixed", they will probably need to be documented.

You'll need this test-code to reproduce everything and the addition of asyncio.new_event_loop in PR #5890:
```
try:
    import uasyncio as asyncio

    mp = True
except:
    import asyncio

    mp = False

_tmp_task = None

async def foo():
    await asyncio.sleep(0)
    print("foo")
    1 / 0  # die

async def wait():
    await asyncio.sleep(0.5)
    print("wait done")

def _handle_exception(loop, context):
    print("exception:", context["exception"] if "exception" in context else context["message"])

def _handle_exception_and_stop(loop, context):
    print("exception and stop:",
          context["exception"] if "exception" in context else context["message"])
    loop.stop()

def test(name, func, *args, stop_after_exception=False):
    print("---------------------------------")
    print("--", name, "stop after exception:", stop_after_exception)
    print("---------------------------------")
    l = asyncio.new_event_loop()
    l.set_exception_handler(
        _handle_exception_and_stop if stop_after_exception else _handle_exception)
    l.create_task(func(*args))
    if mp:
        l.create_task(wait())
        l.run_forever()  # in mp l.run_until_complete doesn't get stopped by loop.stop()
    else:
        try:
            l.run_until_complete(wait())
        except RuntimeError as e:
            print(e)

##############################################

async def exception_is_caught(*args, **kwargs):
    t = asyncio.create_task(foo())
    for _ in range(3):
        print('Working')  # foo is dead, but we carry on
        await asyncio.sleep(0.05)
    try:
        res = await t
    except Exception as e:
        res = e
    print("res", res)
    raise AttributeError("Stop test")  # Stops the scheduler in exception handler

async def exception_uncaught(*args, **kwargs):
    t = asyncio.create_task(foo())
    for _ in range(3):
        print('Working')  # foo is dead, but we carry on
        await asyncio.sleep(0.05)
    raise AttributeError("Stop test")  # Stops the scheduler in exception handler
[... 213 lines cut ...]
exception: divide by zero
Working
Working
exception: Stop test
wait done
---------------------------------
-- Exception_is_NOT_caught_and_task_globally_assigned stop after exception: True
---------------------------------
Working
foo
exception and stop: divide by zero
```
As we can see in the CPython output, the exception is held back and since the global variable doesn't go out of scope, the exception of foo() is actually raised inside the next test, when the new loop starts.

## Summary / tl;dr

In Cpython asyncio exceptions are only raised to the exception handler once the variable to which the task is assigned to goes out of scope (if the task was ever assigned to a variable) AND if that task was never awaited. 
In Upy an exception will always be raised to the exception handler if the exception occurs while no other task is actively awaiting the failing task.

### Why does this difference matter?
If you rely on a custom exception handler you might end up stopping your program at a different state than Cpython or process exceptions which are then caught by another task. Throw some timing variations into the mix and you can end up with a program that will behave differently on each execution (depending on the timing of awaiting the failing task). Sometimes you will get to the exception_handler, sometimes you don't. Depending on what you do with the exception you might have logged exceptions that are actually just fine or, if you decide to terminate your program when an exception occurs, you might terminate your program permaturely without a real problem.

### How can this be fixed?
Sadly I don't see an easy way to fix this, because it doesn't only rely on uasyncio internals. I don't know much about the micropython c(++) workings but I doubt that it'll be easy to determine when exactly a task isn't assigned to any variable anymore. 
So this will probably end up being a CPython difference that needs to be documented but can't (easily) be fixed.

dpgeorge, 2020-04-12:
I did also notice some differences here, and the main thing I noted is that CPython gives the error about unhandled exceptions only when the task is garbage collected / freed via reference counting.  From this point of view it could be argued that it's non-deterministic when (or even if) unhandled exceptions are reported.  So it could be that the only sensible solution is to not report such exceptions at all (and act like there is infinite memory, nothing gets garbage collected).

kevinkk525, 2020-04-12:
I feel that might be a stretch to say it is non-deterministic. 
So far in all my tests I was able to reproduce a deterministic behaviour but I'm sure there might be cases with non-deterministic behaviour.
If it is tied to garbabe collection then the CPython garbage collection seems to work quite deterministic in my tests. The exceptions are always raised to the exception handler when a Task gets destroyed/freed even if other tasks keep running. 

I guess the main question is: How do you want to handle it in upy? 
So far I can see 2 options:
1) Make it like CPython using reference counting
2) Keep the API compatible to CPython but firmly state that it behaves differently from CPython to give better logging in upy. (especially the point about exceptions being raised to the exception handler although they might get caught later might be interesting to document)

dpgeorge, 2020-04-13:
> I feel that might be a stretch to say it is non-deterministic.

If behaviour relies on running an object finaliser then (AFAIK) such behaviour is not specified within the Python language, because a given Python implementation does not need to use reference counting.  If an implementation of Python uses garbage collection only and has a large heap (eg MicroPython unix port) then finalisers may never run (except possibly at program exit).

I haven't studied all the tests in the original post at the top, but a simple test which shows one of the issues (and perhaps all issues can be reduced to this) is:
```python
try:
    import uasyncio as asyncio
except:
    import asyncio

async def task():
    raise ValueError

async def main():
    t = asyncio.create_task(task())
    await asyncio.sleep(0)
    try:
        await t
    except ValueError:
        print('ValueError')

asyncio.run(main())
```
CPython will print:
```
ValueError
```
but MicroPython will print:
```
Task exception wasn't retrieved
future: <Task object at 7f9e17a27de0> coro= <generator object 'task' at 7f9e17a27fc0>
Traceback (most recent call last):
  File "extmod/uasyncio/core.py", line 183, in run_until_complete
  File "test.py", line 7, in task
ValueError: 
ValueError
```
The issue is that the failing task is waited on *after* the main loop has already processed the unhandled exception, not knowing that something will wait on it after.  Apart from printing the spurious error message about an exception not being retrieved, the program will continue to run correctly.

This behaviour should probably be fixed, otherwise there will be spurious error messages.  But I don't think it's possible to fix it without doing like CPython does and printing the error message only in the finaliser when the task is GC'd.

kevinkk525, 2020-04-13:
Yes this test equals "Exception_is_caught" and is the main issue with exception handling.

> If behaviour relies on running an object finaliser then (AFAIK) such behaviour is not specified within the Python language, because a given Python implementation does not need to use reference [...]

I'm not sure this is true for asyncio because the exception handling behaviour needs a deterministic system to be reliable. A finaliser that may or may not run wouldn't be too helpful?

I'd certainly prefer it to be fixed but I also understand that it might be a lot of effort in upy with increased code size and RAM demands.

peterhinch, 2020-04-13:
The behaviour of @dpgeorge's script, while not CPython compliant, seems reasonable. At the moment when the exception occurs, it's unhandled.