micropython/micropython #5913: ESP32 writeblocks with buffer size not multiple of blocksize gives OSError

chtinnes, 2020-04-12:
Hey,

Just realized, that writing data into partition with a data buffer with size != 0 mod BLOCK_SIZE leads to an `OSError: -260`.
It is due to the erase call `esp_partition_erase_range(self->part, offset, bufinfo.len)` in
`esp32_partition_writeblocks`.

I am not sure if this is intended behavior. If not, I would recommend either to return a proper error message or to erase ceil(buffer_size/BLOCK_SIZE) blocks.

dpgeorge, 2020-04-13:
> I am not sure if this is intended behavior.

It is intended behaviour: for block devices (using the standard interface, not extended interface) all reads and writes must be a multiple of the block size.  This is why the location of the read/write is given as block number rather than byte offset.

> I would recommend either to return a proper error message

It' looks like -260 is an internal IDF error code.  Probably that could be converted to a "standard" errno number.

klondi, 2024-01-10:
I was looking for this issue. -260 is Invalid size
```/usr/share/esp-idf/components/esp_common/include/esp_err.h:#define ESP_ERR_INVALID_SIZE        0x104   /*!< Invalid size */```