micropython/micropython #5915: socket.makefile() doesn't take keyword arguments

amotl, 2020-04-12:
Dear Damien,

this is just a heads up for a very minor nit. However, we felt obligated to report it upstream.

While we recognized `socket.makefile()` is a noop on MicroPython, we introduced it into some implementations of the `urequests` and `umqtt` modules to make them compatible with CPython in order to be able to invoke them within a `pytest`-based test harness (see also #5786).

While we found the documentation [1] is saying that passing keyword arguments would be possible, @poesel found out (see https://github.com/hiveeyes/terkin-datalogger/issues/72) that it actually isn't:
```
  File "/lib/terkin/telemetry.py", line 684, in connect
  File "/dist-packages/umqtt.py", line 65, in connect
TypeError: function doesn't take keyword arguments
```

On top of that, passing the `buffering`-parameter as boolean variable also doesn't work. It has to be passed as `0` (zero).

To make the long story short, this works:
```
sock.makefile('rwb', 0)
```

while that doesn't:
```
sock.makefile(mode="rwb", buffering=False)
```

With kind regards,
Andreas.

[1] https://docs.micropython.org/en/latest/library/usocket.html#usocket.socket.makefile

dpgeorge, 2020-04-13:
> While we found the documentation [1] is saying that passing keyword arguments would be possible

The documentation has a `/` at the end of the arguments for `socket.makefile` which means it does not accept keywords.  Not accepting keywords is to save on code size (as you probably know) and in this case I don't see why it's needed to parse keywords, it should be possible to use positional args here.

> On top of that, passing the `buffering`-parameter as boolean variable also doesn't work. It has to be passed as `0` (zero).

According to CPython docs `buffering` should be an integer: https://docs.python.org/3/library/functions.html#open (search for buffering).

But which port are you using?  Because I'm pretty sure none of the ports actually parse the `buffering` argument.

amotl, 2020-04-25:
Dear Damien,

thanks for your thorough answer.

> The documentation has a `/` at the end of the arguments for `socket.makefile` which means it does not accept keywords.

I see. Haven't been aware of that.

> Not accepting keywords is to save on code size [...]

Makes sense, thanks for outlining that!

> Which port are you using?

Actually, we are currently conceiving a test harness for MicroPython code based on [pytest](https://pytest.org/) [1]. As this is invoked on regular CPython, we are even executing the code on no MicroPython port at all.

With kind regards,
Andreas.

P.S.: After some more fiddling regarding compatibility with Pycom's MicroPython [2,3], we are now happy the respective implementations work on all of (Genuine MicroPython, Pycom MicroPython and regular CPython).

[1] https://community.hiveeyes.org/t/developing-terkin-for-micropython/233/25
[2] https://github.com/daq-tools/pycopy-lib/commit/5248ad1033eb5b9d2d65d2d89f3391107e79f4e3
[3] https://github.com/daq-tools/pycopy-lib/commit/a6b3cde8a5df2726ff387141ef90a37d926083dc

closed 2020-04-25 by amotl