micropython/micropython #5916: Start block calculation for second storage blockdevice wrong (?)

hoihu, 2020-04-12:
https://github.com/micropython/micropython/blob/master/ports/stm32/storage.c#L45

shouldn't it read
`#define FLASH_PART2_START_BLOCK (FLASH_PART1_START_BLOCK + MICROPY_HW_BDEV_IOCTL(BDEV_IOCTL_NUM_BLOCKS, 0))`

as I interpret is, the second blockdevice should be appended to the previous blockdevice, so the start block of blockdevice 2 = start block of blockdevice 1 + length of blockdevice *1* ?

dpgeorge, 2020-04-13:
Yes it looks like it is wrong as it currently exists in the code.

Probably the reason this hasn't been fixed is because the board that originally used this feature (PYBD-SFx) has the same size flash for both PART1 and PART2.  And that board no longer uses it, so it's inactive code.

hoihu, 2020-04-13:
> has the same size flash for both PART1 and PART2. 

Yeah, I suspected this. I'll file a PR.

dpgeorge, 2020-04-16:
Fixed by 7942d0b688075a42e78f87152f952ad4c2976f44

closed 2020-04-16 by dpgeorge