micropython/micropython #5924: TypeError: unsupported type for __hash__: 'range'

fizista, 2020-04-14:
Missing property for the primary operator.
```
>>> d = {}
>>> d[range(10)] = 123
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: unsupported type for __hash__: 'range'
```
I think it would be worth implementing it.

Now we can get around this by creating an extra class:
```
class new_range(range):
    def __hash__(self):
        return hash((self.start, self.stop, self.step))
```

dpgeorge, 2020-04-14:
> I think it would be worth implementing it.

Since MicroPython aims to be small, It's not feasible to implement all missing operations.  And being able to hash ranges doesn't seem to be that common (or useful?).  What is the use case for this, is it widely done?

fizista, 2020-04-15:
I used it when I was writing the event schedule. An example:
```
jobs = { 
    range(...): [<callback_1>, <callback_2>, ...] 
    range(...): [<callback_1>, <callback_2>, ...] 
}
```
I don't know if it's that common practice. In fact, I could use a list instead of a dictionary here.

The only problem that this property solves is that the range can be used as keys in the dictionary.

fizista, 2020-04-15:
I checked how much memory it would take to implement the above problem with a list. And if it makes sense __hash__ in range. And there is no clear answer. On the ESP32 port the list implementation gives ~15% less memory (__hash__ doesn't give any sense here). But the PC port and ESP8266 unexpectedly give the opposite result, a dictionary implementation gives ~15% memory savings.

I don't know why there is such a difference between the ports.

Below I include the test code.

```
=== gc.collect()
=== a=gc.mem_alloc()
=== d=[(new_range(i), i) for i in range(50)]
=== gc.collect()
=== b=gc.mem_alloc()
=== b-a
=== del d
=== 
=== gc.collect()
=== a=gc.mem_alloc()
=== d=[(range(i), i) for i in range(50)]
=== gc.collect()
=== b=gc.mem_alloc()
=== b-a
=== del d
=== 
3472
1872
>>> 100-1872*100/3472
46.08295
```

```
import gc

class new_range(range):
    def __hash__(self):
        return hash((self.start, self.stop, self.step))
        
def test(n=50):
    gc.disable()

    gc.collect()
    t1s=gc.mem_alloc()
    d=[(new_range(i), i) for i in range(n)]
    gc.collect()
    t1e=gc.mem_alloc()
    del d
    
    gc.collect()
    t2s=gc.mem_alloc()
    dd=dict([(new_range(i), i) for i in range(n)])
    gc.collect()
    t2e=gc.mem_alloc()
    del dd
    
    list_mem=t1e-t1s
    dict_mem=t2e-t2s
    diff=list_mem-dict_mem
    print('list mem: %s\ndict mem: %s\nlist_mem - dict_mem = %s (%s)' % (list_mem,dict_mem,diff, 100-dict_mem*100/list_mem))
    
    gc.enable()
```
```
ESP8266 port
>>> test()
list mem: 3472
dict mem: 2912
list_mem - dict_mem = 560 (16.1291)

PC port
>>> test()
list mem: 6944
dict mem: 5888
list_mem - dict_mem = 1056 (15.2073732718894)

ESP32 port
>>> test()
list mem: 3472
dict mem: 4032
list_mem - dict_mem = -560 (-16.12904)
```

dpgeorge, 2020-04-16:
Usually the main factor when implementing a new feature is the amount of flash/ROM/executable code size that it takes/costs.  The tool `tools/metrics.py` can be used to compute this, between two different builds, across ports.

Eg for hashing range objects, the following implements a valid hash but it's not very unique:
```diff
diff --git a/py/objrange.c b/py/objrange.c
index 4eed4b941..81583f5fe 100644
--- a/py/objrange.c
+++ b/py/objrange.c
@@ -134,6 +134,15 @@ STATIC mp_obj_t range_unary_op(mp_unary_op_t op, mp_obj_t self_in) {
     switch (op) {
         case MP_UNARY_OP_BOOL:
             return mp_obj_new_bool(len > 0);
+        case MP_UNARY_OP_HASH: {
+            mp_int_t hash = 0;
+            hash += self->start;
+            hash += self->stop;
+            hash += self->step;
+            return MP_OBJ_NEW_SMALL_INT(hash);
+        }
         case MP_UNARY_OP_LEN:
             return MP_OBJ_NEW_SMALL_INT(len);
         default:
```
This costs (in code size):
```
   bare-arm:   +12 +0.018% 
minimal x86:   +44 +0.030% 
   unix x64:   +32 +0.006% 
unix nanbox:  +100 +0.023% 
      stm32:   +12 +0.003% PYBV10
     cc3200:   +16 +0.009% 
    esp8266:   +24 +0.004% GENERIC
      esp32:   +20 +0.001% GENERIC
        nrf:   +12 +0.008% pca10040
       samd:   +12 +0.012% ADAFRUIT_ITSYBITSY_M4_EXPRESS
```

fizista, 2020-04-16:
In fact, the implementation of this __hash__ is not so simple, with the limitations that micropython code gives.

I have not been able to find a solution to the problem of hash uniqueness. 

At python level someone can implement something like that:

```
class new_range(range):
    def __hash__(self):
        return self.start << 64 + self.step << 32 + self.stop
```

Thank you for your time on this issue.

closed 2020-04-16 by fizista

dpgeorge, 2020-04-18:
> I have not been able to find a solution to the problem of hash uniqueness.

Hashes don't need to be unique, the only requirement is that objects that are equal have the same hash (technically all hashes could be 0, but that's not very useful).

fizista, 2020-04-22:
You're right.

Now my code doesn't need __hash__.

In Python implementation I found such a comment:
```
> file: Python-3.8.2/Objects/rangeobject.c
/* Hash function for range objects.  Rough C equivalent of

   if not len(r):
       return hash((len(r), None, None))
   if len(r) == 1:
       return hash((len(r), r.start, None))
   return hash((len(r), r.start, r.step))
*/
```

If you decide to implement this property it may become useful.

And thank you for your work.