micropython/micropython #5927: tests/pyb/adcall.py: fails on NUCLEO_F411RE and hang board

tve, 2020-04-15:
The pyb/adcall.py has hard-coded pins, some of which (A2&A3) overlap with the uart used for the REPL. This causes the test to fail and the board to become unresponsive to pyboard/run-tests.

tve, 2020-04-15:
I suppose the answer is that `--target pyboard` only works with a real pyboard and that the pyb module is untested for stm32 in general?

tve, 2020-04-15:
After skipping the above failing tests plus `pybnative/while.py` (no LED(2) on nucleo board) I do get two more failures. Do these pass on a pyboard?
```
$ diff extmod_uselect_poll*
1,2c1,4
< 0
< True
---
> Traceback (most recent call last):
>   File "<stdin>", line 14, in <module>
> OSError: no available NIC
> CRASH
\ No newline at end of file

$ diff extmod_usocket_udp_nonblock*
1c1,4
< EAGAIN: True
---
> Traceback (most recent call last):
>   File "<stdin>", line 14, in <module>
> OSError: no available NIC
> CRASH
\ No newline at end of file

dpgeorge, 2020-04-16:
> I suppose the answer is that `--target pyboard` only works with a real pyboard and that the pyb module is untested for stm32 in general?

Yes that's correct.  These `tests/pyb/` tests are quite old and were used to verify pyboard operation from the beginning.

Ideally these tests would be split out in to generic `machine` module tests that run on all ports with that module, stm32-generic tests, and pyboard specific tests.  That would make them more useful, to help people making new ports verify their port works.

> After skipping the above failing tests plus `pybnative/while.py` (no LED(2) on nucleo board) I do get two more failures. Do these pass on a pyboard?

They fail on a standard PYBv1.x (as above, no NIC) but pass on a PYBD.