micropython/micropython #5929: MQTT to AWS IoT Core fails with mbedtls_ssl_handshake_error

Aish-Git, 2020-04-16:
mbedtls_ssl_handshake error: -2880
I (14912) wifi:Traceback (most recent call last):
  File "main.py", line 99, in <module>
  File "main.py", line 96, in <module>
  File "main.py", line 69, in mqtt
  File "umqtt/simple.py", l

Fails on the latest firmware but works on any of the 2019 builds

dpgeorge, 2020-04-16:
> mbedtls_ssl_handshake error: -2880

This is `MBEDTLS_ERR_X509_ALLOC_FAILED`, so it's running out of RTOS heap memory.

Aish-Git, 2020-04-16:
Thanks @dpgeorge  - would adding a gc.collect() and checking for gc.free_men() prior to that socket connect help? I am reading certs into variables but assuming those are in the function stack.

tve, 2020-04-16:
Nope, it's the C heap, not the MP heap... Either use a module with PSRAM or reduce mp_task_heap_size by some factor (e.g. 80%) here https://github.com/micropython/micropython/blob/master/ports/esp32/main.c#L103-L104 and rebuild the fw and try that...

Aish-Git, 2020-04-16:
Hi @tve  - noted. Curiously I tried the same code that worked on earlier firmware (20191220) doesn't work on the latest one 20200416. Getting `mbedtls_ssl_handshake error: -10` error. Could it be an SSL wrap_socket issue?

dpgeorge, 2020-04-18:
> uriously I tried the same code that worked on earlier firmware (20191220) doesn't work on the latest one 20200416

That's most likely due to an update to the ESP-IDF, which changes the heap layout and has less RTOS heap available.

Carglglz, 2020-04-19:
Hi @Aish-Git see the [long explanation](https://github.com/micropython/micropython/issues/5543#issuecomment-577824625) in  #5543

Aish-Git, 2020-04-19:
Thanks @Carglglz - incredibly helpful and sorry didn't see that before I logged this one. Let me try some options listed there in terms of a 1096 bit key to keep the cert length small. I am also going to try the recommendation that @tve provided on increasing the C heap size. Will keep this thread posted...

rjbanna, 2020-07-07:
I also received the ssl_handshake_status error. My error log is as the following

`ssl_handshake_status: -256
Cannot connect MQTT: [Errno 5] EIO
Traceback (most recent call last):
  File "main.py", line 236, in <module>
  File "main.py", line 217, in connect_mqtt
  File "umqtt/simple.py", line 62, in connect
OSError: [Errno 5] EIO`

I'm using AWS IoT with umqtt. Sometimes it get connected but sometime it gives me the above error. Can anyone help me with this?

sactre, 2020-09-18:
I make SSL connections and use AmazonIOT and my memory configuration is this.
`size_t mp_task_heap_size = 50 * 1024; //heap_caps_get_largest_free_block(MALLOC_CAP_8BIT);

void *mp_task_heap = malloc(mp_task_heap_size);`

srcnet2, 2020-12-04:
Downgrading to 1.12 (esp32-idf3-20191220-v1.12.bin) fixed it for me. 1.13 must do something different.

eliclement, 2021-01-19:
@Aish-Git did you had any success? I tried to increase the heap size using @tve's suggestions but no luck. I'm using a ESP32 WROVER with PSRAM. Did you managed to increase the C heap?

Aish-Git, 2021-01-20:
@eliclement I never did get to going down that path as I switched over to a ESP Pico board (TinyPico) that has an onboard SRAM so ended up working well for me.

eliclement, 2021-01-20:
@Aish-Git OK thanks. I had to remove the BLE in the build and which freed some heap memory and it was sufficient.

chtinnes, 2021-03-16:
I had similar issues when connecting to RabbitMQ (actually I think that AWS IoT is using RabbitMQ under the hood). I am not sure if you have control over the Ciphers. For me, I controlled it server side and using DHE-RSA-AES128-GCM-SHA256 i worked without any changes to heap. But anyway, more RAM would always be great =) .

Is it possible to restrict the Ciphers Mbedtls is using? Then, I think AWS IoT will also allow using DHE-RSA-AES128-GCM-SHA256 because it is still recommended and you can connect without issues.

the-stanely, 2021-04-18:
> Downgrading to 1.12 (esp32-idf3-20191220-v1.12.bin) fixed it for me. 1.13 must do something different.

Thank you for that, @srcnet2 .  I've been working on this for a while and v1.12 idf3 binary is the only one that's able to reliably connect to AWS using a generic ESP32.

AWS for IoT is the 800 pound gorilla.  Sure would be nice to use a generic ESP32 with the latest binary in an AWS application. Using idf4, "esp32-idf4-20210202-v1.14.bin" results in:

`OSError: (-17168, 'MBEDTLS_ERR_RSA_PRIVATE_FAILED+MBEDTLS_ERR_MPI_ALLOC_FAILED')`

the-stanely, 2021-04-18:
@sactre said...

> I make SSL connections and use AmazonIOT and my memory configuration is this.
> `size_t mp_task_heap_size = 50 * 1024; //heap_caps_get_largest_free_block(MALLOC_CAP_8BIT);
> 
> void *mp_task_heap = malloc(mp_task_heap_size);`

If that's all it takes, it might be of value to make available updated binaries that are capable of connecting to AWS IoT Core using a generic ESP32.

the-stanely, 2021-04-20:
I have good news and I have bad news!  The new version of MicroPython, idf4 v1.15 -- esp32-20210418-v1.15.bin, works with AWS IoT!  YAY!!!

v1.15 does not have mqtt built in so you have to install your own.  That's OK because it's easy to do.

Now for the bad news.  If I drop into REPL a few times while running my tests, eventually I get a WiFi error that looks like this:
```
Traceback (most recent call last):
  File "<stdin>", line 23, in <module>
OSError: Wifi Internal Error
```
I don't know why, and the message doesn't give me much to go on.  I have to machine.reset() to get the WiFi to work again.  A soft boot (^D) doesn't do it.

I will open another issue about this specific WiFi problem.

the-stanely, 2021-04-20:
I can also force my WiFi station mode inactive in my case station.active(False), and that works to reset the WiFi, but then my software has to reconnect every time when coming back from REPL.  Is it supposed to work like this?

ironss, 2021-08-31:
My solution was to make Micropython leave some of the heap for the RTOS, instead of claiming all of free memory for itself. Experimentation showed that
* 2K reserved never worked
* 4K reserved sometimes worked
* 8K reserved always worked
But this probably depends on what other features are available.
        
        diff --git a/ports/esp32/main.c b/ports/esp32/main.c
        index ff6dd6957..0149e8a0b 100644
        --- a/ports/esp32/main.c
        +++ b/ports/esp32/main.c
        @@ -121,6 +121,7 @@ void mp_task(void *pvParameter) {
             #else
             // Allocate the uPy heap using malloc and get the largest available region
             size_t mp_task_heap_size = heap_caps_get_largest_free_block(MALLOC_CAP_8BIT);
        +    mp_task_heap_size -= 8192;  // Reserve 8K for C heap (mbedTLS needs it)
             void *mp_task_heap = malloc(mp_task_heap_size);
             #endif

Does it make sense to make this a configuration option?

pumelo, 2021-09-01:
By default mbedtls only uses internal ram on the esp32 (security concerns). However esp-idf provides a config to allow mbedtls to alloc also in the PSRAM space:  
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-mbedtls-mem-alloc-mode

nickboucart, 2022-10-13:
Solution proposed by @ironss above did the trick for me. After recompiling the latest trunk version of micropython (October 13th 2022), I can now connect to AWS IoT and send and receive messages.

Big thank you @ironss, spent about 3 days before I stumbled upon this solution.

Tangerino, 2022-10-13:
The option described by @pumelo is implemented in latest MP? The "...alloc also in the PSRAM..."

bneigher, 2023-04-12:
How would we go about debugging `MBEDTLS_ERR_SSL_BAD_HS_CLIENT_HELLO` errors?  I have attached my code for reference... on ESP32-C3.  I'm using the latest nightly of [micropython for esp32-c3](https://micropython.org/download/esp32c3-usb/) (esp32c3-usb-20230411-unstable-v1.19.1-1014-gbde222ce8.bin), the latest version of [mqqt_as](https://github.com/peterhinch/micropython-mqtt/tree/master/mqtt_as)

```
import gc
import time
import machine
import network
import json
gc.collect()

import ussl as ssl
import uasyncio as asyncio
from ubinascii import hexlify
gc.collect()

from net import connect # just a wrapper for Wifi connection
from lib.mqtt_as import config
from lib.mqtt_as import MQTTClient
gc.collect()

client_id = hexlify(machine.unique_id()).decode('utf-8')

# Set up the AWS IoT endpoint details
endpoint = "xxxxxxxxxxxxx-yyy.iot.us-west-2.amazonaws.com"
topic = "sdk/test/python"

# Set up the AWS IoT certificate and key files
# cert_file = "/flash/cert/esp32-c3.cert.pem"
cert_file = "/flash/cert/device.cert.der"

# key_file = "/flash/cert/esp32-c3.private.key"
key_file = "/flash/cert/device.private.der"

# root_ca_file = "/flash/cert/root-CA.crt"  # Path to the root CA certificate file
root_ca_file = "/flash/cert/root-CA.der"  # Path to the root CA certificate file

PVT_KEY = open(key_file,'rb').read()
CERT_KEY = open(cert_file, 'rb').read()
CA_KEY = open(root_ca_file, 'rb').read()

ssl_params = {
    'key': PVT_KEY,
    'cert': CA_KEY,
    'server_side': True,
    'server_hostname': endpoint,
    'do_handshake': True,
    'cert_reqs': ssl.CERT_REQUIRED,
    'cadata': CA_KEY
}
                    
# Connect to AWS IoT MQTT
def connectMQTT():
    internet_connection = connect()
    print("MQTT CONNECTING", internet_connection)
    config['client_id'] = client_id
    config['server'] = endpoint
    config['port'] = 8883
    config['keepalive'] = 5000
    config['ssl'] = True
    config['ssl_params'] = ssl_params
    config['ssid'] = internet_connection['ssid']
    config['wifi_pw'] = internet_connection['password']
    MQTTClient.DEBUG = True
    client = MQTTClient(config)
  
    try:
        asyncio.run(client.connect(quick=True))
        print("MQTT CONNECTED") #, client.isconnected())
    finally:
        print("MQTT CLOSED") #, client.isconnected())
        # client.close()
        # client.disconnect()
    
    return client

    
# Publish a message to the topic
def sendMQTT(client, message):
    print(f"Sending: {client.isconnected()} {message}")
    client.publish(topic, message)

connectMQTT()
```

Tried the stable micropython as well, doesn't seem to make any difference.

```
Connecting to broker.
Traceback (most recent call last):
  File "<stdin>", line 9, in <module>
  File "util.py", line 87, in <module>
  File "util.py", line 70, in connectMQTT
  File "uasyncio/core.py", line 1, in run
  File "uasyncio/core.py", line 1, in run_until_complete
  File "uasyncio/core.py", line 1, in run_until_complete
  File "lib/mqtt_as.py", line 637, in connect
  File "lib/mqtt_as.py", line 271, in _connect
  File "lib/ssl.py", line 30, in wrap_socket
OSError: (-30976, 'MBEDTLS_ERR_SSL_BAD_HS_CLIENT_HELLO')
```

MKoushikYadav, 2023-05-13:
> 

Hello, @ironss. How would you go about doing this on micropython. Should I edit the binary files or something? Its still 2023 and none of the solutions work so far. I am trying out different versions to test  stuff.