micropython/micropython #5932: RFC: Should we add capitalization to first word in error strings?

hardkrash, 2020-04-16:
With the transition to compressed strings we lost capitalization of the first word in error strings as a normalization.  A great compromise for space savings, but a bit of lost polish.

Could we get that back and have the system work?
Is it safe to assume that compressed error strings could always capitalize as well, e.g. subtract 32 from the first byte if  97 <= first byte <= 122?

In the code we only add spaces before a word if it is not the first word. i.e. not the 0->1 transition.
https://github.com/micropython/micropython/blob/073b9a5eb888b26be7ce4c65e9649b6c3857ba39/py/qstr.c#L355

tve, 2020-04-17:
Personally I really like the unix convention of not capitalizing error strings because they often get embedded in other messages where the capitalization ends up out of place.

dpgeorge, 2020-04-17:
> With the transition to compressed strings we lost capitalization of the first word in error strings as a normalization.

I'm not sure why you have this impression, the standard has mostly been to have lowercase for the first word, following most of CPython's error messages.  Eg in CPython:
```
>>> 1/0
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ZeroDivisionError: division by zero
```

hardkrash, 2020-04-17:
Because I’m reading the commits out of context, and didn’t realize that the message is prepended with the exception type.  Sorry for the noise.

closed 2020-04-17 by hardkrash