micropython/micropython #5934: Spi ram fails to initialize during boot - esp32

bnlerner, 2020-04-17:
### FW Version
both idf3 and 4
GENERIC-SPIRAM : esp32spiram-idf3-20200417-v1.12-375-g28833690b.bin

GENERIC-SPIRAM : esp32spiram-idf4-20200417-v1.12-375-g28833690b.bin

### HW Version
[Espressif ESP32 Development Board - Developer Edition](https://www.adafruit.com/product/3269)

### Error message during boot up

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5008
ho 0 tail 12 room 4
load:0x40078000,len:11608
ho 0 tail 12 room 4
load:0x40080400,len:6120
entry 0x400806bc
I (605) psram: This chip is ESP32-D0WD
E (605) spiram: SPI RAM enabled but initialization failed. Bailing out.
I (606) cpu_start: Failed to init external RAM; continuing without it.
I (613) cpu_start: Pro cpu up.
I (616) cpu_start: Application information:
I (621) cpu_start: Compile time:     Apr 17 2020 00:42:31
I (627) cpu_start: ELF file SHA256:  0000000000000000...
I (633) cpu_start: ESP-IDF:          v3.3.2
I (638) cpu_start: Starting app cpu, entry point is 0x40083d70
I (0) cpu_start: App cpu up.
I (649) heap_init: Initializing. RAM available for dynamic allocation:
I (656) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (662) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (668) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (674) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (680) heap_init: At 3FFCAB78 len 00015488 (85 KiB): DRAM
I (686) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (692) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (699) heap_init: At 4009A528 len 00005AD8 (22 KiB): IRAM
I (705) cpu_start: Pro cpu start user code
I (53) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
E (40) spiram: SPI RAM not initialized

tve, 2020-04-17:
Are you sure yours has an SPIRAM? The one pictured on the page you linked to doesn't... If your esp32 module says WROOM you don't have SPIRAM, if it says WROVER then you do.

bnlerner, 2020-04-17:
sorry my mistake. ill close

closed 2020-04-17 by bnlerner