micropython/micropython #5935: Add ARMV5T (LE) support

martmists, 2020-04-17:


tve, 2020-04-17:
Did you want to make a PR? ;-)

dlech, 2020-04-17:
I've been running the unix port on ARMv5t for several years.

martmists, 2020-04-17:
@dlech Doesn't that require an unix-like system? I'm trying to embed micropython on some old game consoles, so I doubt the unix port would work for me.

dlech, 2020-04-17:
The core runtime is the same, so we know it compiles/works on that architecture. So it should be possible to start with the minimal port or the bare-arm port and change the compiler options to target your architecture.

martmists, 2020-04-17:
The latest release doesn't offer armv5t in mpy-cross though, how would I solve this?

dlech, 2020-04-17:
just emit mpy bytecodes instead of native code?

martmists, 2020-04-18:
But then how do I get mpy running on armv5t? From what I can tell from the documentation, I need to compile it with mpy-cross to get the object file.

dpgeorge, 2020-04-18:
> But then how do I get mpy running on armv5t?

It depends exactly what you want to do.  For most things you don't need specific machine/architecture support, and MicroPython should run just fin on armv5t.  It's only if you want to use the native emitter, or dynamically loadable native modules, that it needs special architecture support.

jonnor, 2024-07-03:
Is this still relevant?