micropython/micropython #5937: Support for Arduino Due board

whatnick, 2020-04-18:
There is a fairly dated fork with a port for SAM3X processor used in the Arduino Due : https://github.com/Adirockzz95/micropython . Can this be brought in line with upstream work ? at least reusable part moved into appropriate subfolder ?

I can assist once I have checked the original port still works.

jonnor, 2024-10-01:
The Arduino core for Due has not been updated for 4 years now. https://github.com/arduino/ArduinoCore-sam
A post in the forum from 2020 suggests that development has stalled, https://forum.arduino.cc/t/due-development-stalled/938982/5

projectgus, 2025-02-04:
I'm going to close this out for the reasons mention above. If someone is keen to work on an upstreamable port for the ATSAM3X then suggest opening a new [Discussion](https://github.com/orgs/micropython/discussions/) to talk through the details.

closed 2025-02-04 by projectgus