micropython/micropython #5938: MicroPython website no longer offers daily builds

peterhinch, 2020-04-18:
Is this deliberate? It seems unfortunate if users can't try the latest features without compiling. See [this forum post](https://forum.micropython.org/viewtopic.php?f=2&t=8190).

kevinkk525, 2020-04-18:
I agree, I was relying on those daily builds for esp32! 
(Getting esp32 to build locally is a pain so using those was really good)

peterhinch, 2020-04-19:
I also use them for ESP32. FWIW I think the new layout is great for new users. 

There should be a link to a page offering all the build options. People use old release builds to check for regressions. Some lost souls use threading instead of following the path to enlightenment that is uasyncio...

peterhinch, 2020-04-19:
This has now been fixed.

dpgeorge, 2020-04-19:
This still looks broken...

dpgeorge, 2020-04-19:
Ok, should now be fixed, daily builds appear again.  The problem was the build server was updated, now uses Python 3.8, and the old virtual env for Python 3.7 stopped working.

closed 2020-04-19 by dpgeorge