micropython/micropython #5940: Flushing btree database results in panic

carstenblt, 2020-04-18:
I am getting some non-deterministic panics when writing the content of my btree database to flash (`flush()`).

As far as I can tell:
- it doesn't correlate with the amount of data written
- at some point it was deterministic when I had quite a few .mpy modules imported (around 20 I guess). I consolidated a few, and it didn't happen anymore. Now I see it appear randomly, and I can't reproduce it by adding some dump modules.

I would appreciate any suggestions, even for workarounds...

```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400eb080  PS      : 0x00060e30  A0      : 0x801c6071  A1      : 0x3ffcff70
A2      : 0x3f81c410  A3      : 0x00001000  A4      : 0x00000000  A5      : 0x3f40b0d8
A6      : 0x3f866c4c  A7      : 0x00000004  A8      : 0x00000000  A9      : 0x3ffd0080
A10     : 0x3f81c410  A11     : 0x00000002  A12     : 0x3ffcff70  A13     : 0x3ffc20b4
A14     : 0x00000000  A15     : 0x3f4081b4  SAR     : 0x00000006  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000008  LBEG    : 0x4009472e  LEND    : 0x40094739  LCOUNT  : 0x00000000

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

Backtrace: 0x400eb07d:0x3ffcff70 0x401c606e:0x3ffcffa0 0x40101965:0x3ffcffc0 0x400feecc:0x3ffcffe0 0x400fecf1:0x3ffd0020 0x400e4635:0x3ffd0040 0x400e0431:0x3ffd0060 0x400e057e:0x3ffd0080 0x400ee909:0x3ffd00a0 0x400e4ae3:0x3ffd0140 0x400e4b4d:0x3ffd0160 0x400e4bbc:0x3ffd0190 0x400e4671:0x3ffd01b0 0x400e0431:0x3ffd01d0 0x400e057e:0x3ffd01f0 0x400ee909:0x3ffd0210 0x400e47da:0x3ffd02b0 0x400e0431:0x3ffd02e0 0x400ee88d:0x3ffd0300 0x400e47da:0x3ffd03a0 0x400e0431:0x3ffd03f0 0x400e057e:0x3ffd0410 0x400ee909:0x3ffd0430 0x400e47da:0x3ffd04d0 0x400e0431:0x3ffd0530 0x400e045a:0x3ffd0550 0x40103296:0x3ffd0570 0x401034dd:0x3ffd0600 0x4010353b:0x3ffd0620 0x400f67b7:0x3ffd0640 0x40094c1d:0x3ffd0670
```

carstenblt, 2020-04-18:
Also it seems to be tied to btree. Put a `open('test', 'w').write('test')` before the flush and it didn't crash with that write.

tve, 2020-04-18:
It's a bit difficult to help you without any context, such as:
- MP version
- Filesystem type
- Calls that cause the issue
- Code to reproduce
- An actual backtrace

carstenblt, 2020-04-18:
- current head
- now littlefs2, but the behaviour was the same with fat
- can be as simple as
```
f = open('database', "w+b")
database = btree.open(f)
database['test'] = 'test'
database.flush() # crash
```
Reproduction is difficult and probably is related to memory usage...

Is there a tool to follow the backtrace?

tve, 2020-04-18:
> Is there a tool to follow the backtrace?

Build the firmware with DEBUG=1 on the make commandline (need the -g GCC option).
Run the shell script in https://github.com/tve/esp32-backtrace and paste the meditation error block.

carstenblt, 2020-04-18:
Ah, thank you!

```
BT-0: 0x400eafdc is in mp_stream_posix_lseek (../../py/stream.c:534).
534	    const mp_stream_p_t *stream_p = o->type->protocol;
BT-1: 0x401c5ed6 is in mpool_write (../../lib/berkeley-db-1.xx/mpool/mpool.c:370).
370		if (mp->fvtable->lseek(mp->fd, off, SEEK_SET) != off)
BT-2: 0x40101841 is in mpool_sync (../../lib/berkeley-db-1.xx/mpool/mpool.c:284).
284			if (bp->flags & MPOOL_DIRTY &&
BT-3: 0x400fedae is in __bt_sync (../../lib/berkeley-db-1.xx/btree/bt_close.c:178).
178		return (RET_SUCCESS);
BT-4: 0x400febd1 is in btree_flush (../../extmod/modbtree.c:86).
86	    return MP_OBJ_NEW_SMALL_INT(__bt_sync(self->db, 0));
BT-5: 0x400e45c2 is in fun_builtin_1_call (../../py/objfun.c:70).
70	    return self->fun._1(args[0]);
```

carstenblt, 2020-04-18:
So I got some weird behavior, which is probably the same problem. My setup is as follows: I have a module with a "_Config" class, that handles storing of some data with btree. The module defines the class and creates an instance `Config = _Config()`.

Now
```
import config
import gc
gc.collect()
config.Config['some_key'] = 'some value'
```
always results in a `ValueError` exception by the `flush()` call. Without the garbage collection in between it succeeds. It seems like garbage collection is the problem. A btree instance at REPL does not show this behavior.

dpgeorge, 2020-04-19:
> Without the garbage collection in between it succeeds. It seems like garbage collection is the problem.

Actually, yes, that looks to be the problem: the btree library uses `malloc()` to allocate the main DB structure, which stores a pointer to the underlying file object.  On esp32 this goes to the system malloc which is not scanned for root pointers during a GC.  So the file is freed and the error occurs.

A workaround (as you found out above) is to retain a pointer/reference to the file object.

The proper fix would be for btree to use the MicroPython memory allocation function, not the esp32 RTOS one.

dpgeorge, 2020-04-20:
The unix port has the same problem.

dpgeorge, 2020-04-20:
I proposed a fix in #5946, but didn't yet test it against the scenario here.

carstenblt, 2020-04-21:
It works perfectly. Thanks both of you for the help and the quick fix! You are awesome!

closed 2020-04-21 by carstenblt

dpgeorge, 2020-05-02:
Fixed by commit 73c58150f53d9d828c4fc8fb455cca6831eb8ddd