micropython/micropython #5947: Feature request: multiple event loops

traverseda, 2020-04-20:
I think this needs to go here and not in the micropython-libs repo...

A project I'm working on has two main contexts code can run in. Previously I had modeled those two contexts as two different event loops, with an explicit hand-off between them.

One loop was the "motion control" event loop. It was a higher priority and there were only specific spots it would be able to stop at safely. I could turn it into a blocking function (with the help of interrupts to handle some edge cases) but there are reasons why it's better for it to be co-operative multi-tasking based. The idea was that it would call motionLoop.stop() at regular intervals to allow the general loop to run.

The other loop was the general purpose loop which would run things like user plugins, a web interface, etc.

Any thoughts?

peterhinch, 2020-04-21:
Why can't you have a "general purpose" task which would be created at regular intervals which runs these things?

If you need the "main loop" to stop while this is going on, this could be done with an **Event**, set at the start of  "general purpose" and cleared at the end. "Main loop" tasks would pause on the **Event**.

There may be a more elegant solution - it's hard to say without a better understanding of your system. Perhaps the **Barrier** class in my repo might help?

traverseda, 2020-04-21:
There are absolutely ways to work around this, just most of them are equivalent to implementing something that behaves very much like two separate event loops. I can swap out all the ~tasks~ coroutines in the event loop, basically maintain two separate sets of tasks, and manually swap them. I could also turn my motion control loop into a blocking function, essentially implementing my own event loop system to deal with it.

Like I say, there are ways to get around this, but two different event loops is in my opinion a more elegant and readable solution. It works in CPython and I'm not seeing any fundamental reason why it couldn't work in micropython. It's a very elegant way to mix hard(ish)-real-time and soft-real-time requirements.

dpgeorge, 2020-04-22:
It should be technically possible to add support for multiple event loops, it's just a question of how much code would need to be changed and whether it could be made optional.  If most uses of uasyncio don't need multiple event loops then it's hard to argue that it should be a core feature.

It'd be good to implement a proof-of-concept and see how much interest the feature gets.

peterhinch, 2020-04-22:
As one who's used cooperative scheduling in firmware for nearly 40 years I've never wanted it.

traverseda, 2020-04-22:
@peterhinch how would you recommend having motion control and a web interface running on the same board? In my particular case the motion control loop would be a bit more complicated, as it would be getting information on things like missed steps and load-measurement from trinamic stepper drivers. My plan was for the motion control loop to do that kind of book-keeping, polling the stepper drivers for more information, as an asyncio task. Driving the motors would also be an asyncio task. The goal here is to provide a motion control system that lets you move along a particular gcode path while applying a particular amount of *force/load* to an object. The intent being to make CNC milling easier.

I'd like that firmware to also be able to run user code and a web interface.

My thought was that two different event loops, a "private" event loop for code that needs to meet soft real time requirements and a public event loop that user plugins could run in.

Any advice?

peterhinch, 2020-04-23:
Please could you post this query in [the forum](http://forum.micropython.org/index.php) to reduce noise in this bug tracker. I will respond and it will reach more people who may contribute ideas.

RetiredWizard, 2023-02-10:
Sorry about the PR link, it was meant to be a CircuitPython link. I've removed the erroneous one from my PR