micropython/micropython #5948: SPI-3 Wire and UART 1-Wire Support

kwagyeman, 2020-04-20:
I'm working on a new interface library for the OpenMV Cam and I thought these would be nice to haves:

* Add support to the SPI bus to share MISO/MOSI. This feature is supported by the STM32 hardware.
* Add support to the UART to share TX/RX. This feature is also supported by the STM32 hardware.
* Add support to the UART for driver enable for rs485. This feature is also supported by the STM32 hardware.

Supporting these features involves setting a bit in the config register and then just configuring the I/O pins in slightly different modes.

dpgeorge, 2020-04-22:
Adding support for these sounds good to me.  They all sound like one-time configuration options so would go in the constructor/init method.

For the sharing of MISO/MOSI and TX/RX this could be accommodated by specifying the same pins for `mosi`/`miso` and `tx`/`rx` when constructing the object, eg:
```python
spi = SPI(1, mosi=Pin('X7'), miso=Pin('X7'))
uart = UART(1, tx=Pin('X1'), rx=Pin('X1'))
```
The alternative would be a new keyword argument, eg `bidirectional=True`.

For RS485, that should probably just be a keyword argument in the constructor.

Also important to think how other ports can support this, eg esp32.

kwagyeman, 2020-04-22:
Okay, I might do the code change myself and send an upstream PR. I'm doing an interface library for the OpenMV Cam and I'm working through all the bugs in these interfaces and using them in a high speed manner. For example, I've got RAW image transfer over SPI working at 20 MHz moving 1.5MB per second without glitches.