micropython/micropython #5949: Heap size fluctuations

ZuluSpl0it, 2020-04-20:
I’m using MicroPython 1.12 for the ESP8266 target. I’ve noticed weird Heap size changes while using the REPL. For example, from a fresh soft reboot:

>>> import gc
import micropython
micropython.mem_info()
stack: 2080 out of 8192
GC: total: 37952, used: 7760, free: 30192
 No. of 1-blocks: 45, 2-blocks: 9, max blk sz: 264, max free sz: 788

I create an object just to allocate space on the heap:
>>> dummyObj_1 = ("a" * 3000)

I run the micropython.mem_info() command again:
>>> micropython.mem_info()
stack: 2096 out of 8192
GC: total: 37952, used: 16784, free: 21168
 No. of 1-blocks: 40, 2-blocks: 9, max blk sz: 264, max free sz: 582

Heap size shows 16784 used. If I continue to issue micropython.mem_info() I get the heap used values that are about the same (16-17k) .
>>> micropython.mem_info()
stack: 2096 out of 8192
GC: total: 37952, used: 16800, free: 21152
 No. of 1-blocks: 41, 2-blocks: 9, max blk sz: 264, max free sz: 394

>>> micropython.mem_info()
stack: 2096 out of 8192
GC: total: 37952, used: 16912, free: 21040
 No. of 1-blocks: 48, 2-blocks: 9, max blk sz: 264, max free sz: 599

However, every fourth time I issue the command I get:
>>> micropython.mem_info()
stack: 2096 out of 8192
GC: total: 37952, used: 9808, free: 28144
 No. of 1-blocks: 19, 2-blocks: 4, max blk sz: 264, max free sz: 787

The heap size drops to about 9808. The weird thing is that if I continue to issue micropython.mem_info() I get heap sizes of about 16-17k three times in a row and then the fourth time will be about 9800.

This repeats indefinitely. Every fourth result is about 9800. Timing doesn’t seem to matter. I’ve issued the commands quickly as well as slowly. Still the same behavior.

jimmo, 2020-04-21:
It looks like you might have a `main.py` that is at least doing something (as evident by the memory used on the first call to mem_info()).

It would be quite useful to see the output of `micropython.mem_info(1)` straight after the soft reset.

Here's the equivalent output for my ESP8266 running release 1.12.

```
MPY: soft reboot
MicroPython v1.12 on 2019-12-20; ESP module with ESP8266
Type "help()" for more information.
>>> import gc
>>> import micropython
>>> micropython.mem_info()
stack: 2112 out of 8192
GC: total: 37952, used: 5072, free: 32880
 No. of 1-blocks: 18, 2-blocks: 10, max blk sz: 264, max free sz: 2044
>>> dummyObj_1 = ("a" * 3000)
>>> micropython.mem_info()
stack: 2112 out of 8192
GC: total: 37952, used: 8512, free: 29440
 No. of 1-blocks: 23, 2-blocks: 14, max blk sz: 264, max free sz: 1808
>>> micropython.mem_info()
stack: 2112 out of 8192
GC: total: 37952, used: 8240, free: 29712
 No. of 1-blocks: 14, 2-blocks: 10, max blk sz: 264, max free sz: 1808
>>> micropython.mem_info()
stack: 2112 out of 8192
GC: total: 37952, used: 8304, free: 29648
 No. of 1-blocks: 16, 2-blocks: 11, max blk sz: 264, max free sz: 1808
>>> micropython.mem_info()
stack: 2112 out of 8192
GC: total: 37952, used: 8368, free: 29584
 No. of 1-blocks: 18, 2-blocks: 12, max blk sz: 264, max free sz: 1808
>>> micropython.mem_info()
stack: 2112 out of 8192
GC: total: 37952, used: 8432, free: 29520
 No. of 1-blocks: 20, 2-blocks: 13, max blk sz: 264, max free sz: 1808
>>> micropython.mem_info()
stack: 2112 out of 8192
GC: total: 37952, used: 8496, free: 29456
 No. of 1-blocks: 22, 2-blocks: 14, max blk sz: 264, max free sz: 1808
>>> micropython.mem_info()
stack: 2112 out of 8192
GC: total: 37952, used: 8560, free: 29392
 No. of 1-blocks: 24, 2-blocks: 15, max blk sz: 264, max free sz: 1808
>>> micropython.mem_info()
stack: 2112 out of 8192
GC: total: 37952, used: 8624, free: 29328
 No. of 1-blocks: 26, 2-blocks: 16, max blk sz: 264, max free sz: 1808
```

robert-hh, 2020-04-21:
That all looks like the intended behavior. When you execute commands in REPL or otherwise execute Python scripts. The free memory will fluctuate. For instance, every execution of a REPL command will run the parse_compile_execute steps, which allocate new memory from the heap. Then the free memoroy drops, until eventually the automatic garbage collection is done, and free memory increases again. You could also call manually gc.collect(), for instance by executing:

gc_collect();micropython.mem_info()

in a row. Then you should have almost stable value, just fluctuating by a few bytes.

ZuluSpl0it, 2020-04-21:
Issuing gc.collect();micropython.mem_info() does make it a more consistent allocation. I thought it was weird because the heap fluctuated by 60% and the value dropped exactly every fourth command. I repeated it many times.

I was initially trying to figure out how much heap was needed for a certain size object. In this example, the object was created using "a" * 3000. I was expecting the heap allocation to go up by about 3000. So, from about 7800 to about 10800. But, it actually went up by 9000 and then remained constant for a while. How can I go about figuring out the impact a particular object will have on the heap? It seems like it might be sort of unpredictable. However, that might be a totally incorrect assessment since I'm fairly new to micropython.

robert-hh, 2020-04-21:
"a"*3000 could be a complex operation under the hood. bytearray(3000) should provide something more close to what you expect. You also could try:

gc_collect();b="a"*3000;gc.collect();micropython.mem_info()
 
That returns on the board here an additional usage of 3104 bytes.

stinos, 2020-04-21:
> How can I go about figuring out the impact a particular object will have on the heap? It seems like it might be sort of unpredictable.

Sort of. See https://forum.micropython.org/viewtopic.php?f=2&t=8098.

ZuluSpl0it, 2020-04-21:
@stinos, that was an interesting read. Thank you for pointing it out to me. I'll close this issue, as there is no real issue.

ZuluSpl0it, 2020-04-21:
@stinos, that was an interesting read. Thank you for pointing it out to me.