micropython/micropython #5951: extmod/uasyncio is missing from the ESP8266 manifest

peterhinch, 2020-04-21:
Is this deliberate or an oversight? Including it works fine here.

mcauser, 2020-04-21:
Moved to GENERIC build [here](https://github.com/micropython/micropython/commit/1d4d688b3b251120f5827a3605ec232d977eaa0f), possibly doesn't fit on 1M and 512K boards?

dpgeorge, 2020-04-22:
It is enabled for boards with 2M+ flash.  It could still fit on a 1M board, maybe a stripped-down version (eg without wait_for/gather).

peterhinch, 2020-04-22:
Am I missing something here? If I issue `make` (as per the README) it seems to pick up [this manifest](https://github.com/micropython/micropython/blob/master/ports/esp8266/boards/manifest.py). I guess (thanks Mike) I should issue
```
make BOARD=GENERIC
```
My point here is that a user following the README should get the library.

dpgeorge, 2020-04-22:
Running `make` by itself should default to the `GENERIC` board.  This boards uses a custom manifest specified by `boards/GENERIC/mpconfigboard.mk`.  The manifest is `boards/GENERIC/manifest.py` which includes `boards/manifest.py` and then adds `uasyncio`.

peterhinch, 2020-04-22:
Thank you - I now see where I was going wrong. Sorry for the noise.

closed 2020-04-22 by peterhinch