micropython/micropython #5952: Cannot resolve domains for local/reserved IPv4 ranges (socket.geaddrinfo() returns empty list)

freddyli, 2020-04-21:
I am trying to user MicroPython on an ESP32. When resolving hostnames I get a weird behavior where resolution for private IPv4 ranges is not working.

### How to test

In the REPL:

```python
>>> import usocket as socket
>>> socket.getaddrinfo('192.168.0.1.nip.io', 80)
[]
>>> socket.getaddrinfo('200.168.0.1.nip.io', 80)
[(2, 1, 0, '200.168.0.1.nip.io', ('200.168.0.1', 80))]
```

### Expected result

Resolving the host `192.168.0.1.nip.io` should yield a result.

### Actual result

Resolving the host `192.168.0.1.nip.io` returns an empty list.

### Question

Is this expected behavior?

tve, 2020-04-21:
Interesting...
```
>>> socket.getaddrinfo('192.168.0.1.nip.io', 80)
[]
>>> socket.getaddrinfo('www.google.com', 80)
[(2, 1, 0, 'www.google.com', ('172.217.17.228', 80))]
>>> socket.getaddrinfo('172.217.5.196.nip.io', 80)
[(2, 1, 0, '172.217.5.196.nip.io', ('172.217.5.196', 80))]
```
the clue lies in my DNS server's logs:
```
Tue Apr 21 11:20:34 2020 [1587493234.925] daemon.warn dnsmasq[5159]: possible DNS-rebind attack detected: 192-168-3-1.nip.io
```
Dunno whether you go through a DNS server that has that protection and I'll let you google DNS-rebind attack...

freddyli, 2020-04-21:
Yep! That was it! Thank you so much for this tip @tve !!

The ESP32 was using a different OpenWRT-based router than my laptop - and there was DNS rebind protection enabled... one never stops learning something new =)

closed 2020-04-21 by freddyli