micropython/micropython #5956: Bug in OrderDict

fizista, 2020-04-22:
The error is certainly present in the ports of ESP32, ESP8266 and unix(MicroPython v1.12 on 2019-12-23;)

The code that causes the error:
```
import ucollections
d = ucollections.OrderedDict()
d[5]=5
d[4]=4
d.popitem()
d[3]=3
```
```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400e0ed8  PS      : 0x00060930  A0      : 0x800e09c9  A1      : 0x3ffd05a0  
A2      : 0x00000000  A3      : 0x00000007  A4      : 0x00000004  A5      : 0x00000000  
A6      : 0x0000002a  A7      : 0x3ffd0750  A8      : 0x00000003  A9      : 0x00000000  
A10     : 0xc0bf722c  A11     : 0x00000001  A12     : 0x00060920  A13     : 0x00000000  
A14     : 0x0000000a  A15     : 0x3ffd0730  SAR     : 0x00000020  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000004  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0xffffffff  

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

Backtrace: 0x400e0ed5:0x3ffd05a0 0x400e09c6:0x3ffd05c0 0x400e2971:0x3ffd05f0 0x400e29c2:0x3ffd0610 0x400e130b:0x3ffd0630 0x400ec465:0x3ffd0650 0x400e3648:0x3ffd06f0 0x400df471:0x3ffd0750 0x400df49e:0x3ffd0770 0x401038f8:0x3ffd0790 0x40103b44:0x3ffd0830 0x400f54b0:0x3ffd0870 0x400953c1:0x3ffd08a0
```

jimmo, 2020-04-22:
Can repro on Unix -- looks like maybe related to the new equality-checking changes.

```
>>> import collections
>>> d = collections.OrderedDict()
>>> d[5] = 5
>>> d[4] = 4
>>> d.pop
pop             popitem
>>> d.pop
pop             popitem
>>> d.popitem()
(5, 5)
>>> d[3] = 3

Program received signal SIGSEGV, Segmentation fault.
0x00005555555952b9 in mp_obj_equal_not_equal (op=MP_BINARY_OP_EQUAL, o1=0x4, o2=0x7)
    at ../../py/obj.c:217
217	    if (mp_obj_is_str(o1)) {
(gdb) bt
#0  0x00005555555952b9 in mp_obj_equal_not_equal (op=MP_BINARY_OP_EQUAL, o1=0x4, o2=0x7)
    at ../../py/obj.c:217
#1  0x0000555555595543 in mp_obj_equal (o1=0x4, o2=0x7) at ../../py/obj.c:287
#2  0x0000555555594382 in mp_map_lookup (map=0x7ffff7a30868, index=0x7, 
    lookup_kind=MP_MAP_LOOKUP_ADD_IF_NOT_FOUND) at ../../py/map.c:172
#3  0x000055555559a632 in mp_obj_dict_store (self_in=0x7ffff7a30860, key=0x7, value=0x7)
    at ../../py/objdict.c:611
#4  0x0000555555599a9b in dict_subscr (self_in=0x7ffff7a30860, index=0x7, value=0x7)
    at ../../py/objdict.c:203
#5  0x0000555555595ddb in mp_obj_subscr (base=0x7ffff7a30860, index=0x7, value=0x7)
    at ../../py/obj.c:530
#6  0x00005555555b606a in mp_execute_bytecode (code_state=0x7fffffffd7b0, inject_exc=<optimized out>)
    at ../../py/vm.c:530
```

jimmo, 2020-04-22:
Nope, not that at all.

popitem just wasn't implemented for ordered dicts and the non-ordered behavior left the dict in a bad state. See #5957 for a proposed fix.