micropython/micropython #5958: ESP32-S2-Saola-1 Dev board

palmtreefrb, 2020-04-22:
Just receive a couple of these today, the wrover version with 2 MB PSRAM. Any advice on what I can try to compile a micropython to this single core esp32?

jimmo, 2020-04-22:
It's quite a different chip to the ESP32, and currently only supported by an unreleased version on the Espressif toolchain.

MicroPython support will take a fair bit of work, first to update the supported IDF version to 4.2, then changes to supported peripherals etc.

If you're interested in looking into it, I'd start with getting the 4.2 IDF working and test that the examples can deploy to your new boards.

MicroPython uses the IDF in a bit of an unconventional way, essentially bypassing the IDF's build system (formerly Make, now CMake), so this will need to be updated.

Look at #5107 for a recent example of moving from IDF 3.3 to 4.0.

palmtreefrb, 2020-04-23:
Ok, thanks for that info... 
CircuitPython has made their ESP32-S2 port a top priority,  will there be any collaboration with them?

tannewt, 2020-04-23:
@palmtreefrb I'm the lead for CircuitPython and am the person working on ESP32-S2 when I have the time. I'm frequently talking with @jimmo and @dpgeorge. I don't expect to make a PR to MicroPython for support because the two are pretty different. However, all of our code is open source and I'm more than happy to help folks understand our CircuitPython code so they can adapt it to MicroPython. (This is our approach in general.)

teuler, 2020-12-01:
bump?
Actually, I'd really like to see MicroPython for the ESP32-S2. I was playing around with CircuitPython on the FeatherS2 and I must say I am not a huge friend of the drive concept. Under Windows (10) it can be really messy. On one of my PCs (Win 10), the board frequently crashes when attached because Windows does "something" to the drive causing it to reboot. On another PC (Win 10), it works mostly fine. Also, what I really like about the MicroPython way with the pymakr plug-in for Atom is that one can basically work (syncrhonize) from a GitHub repository; with the drive that seems always a bit more complicated ...
In my opinion.

peterhinch, 2020-12-01:
Devices which appear as USB flash drives suffer from [this problem](https://github.com/peterhinch/micropython-samples/blob/master/README.md#15-pyboard-usb-pitfall). On the Pyboard you can disable MSC mode. Can you do this on the FeatherS2?

teuler, 2020-12-07:
Not that I know of.

emard, 2020-12-23:
Is there forseen a micropython version for ESP32-S2.
I'd like it to have 2 usb-serial ports, one for REPL, other for usb-serial to
RX/TX pins. usb-storage? no thanx

sfranzyshen, 2021-01-17:
Should we just switch to CircuitPython If ESP32-S2 support is desired ... or any of the newer hardware devices being released?

amckay1, 2021-02-06:
I'd also like to know whether I should bite the bullet and use circuitpython or hold out for my preferred choice of micropython...

nubcore, 2023-10-30:
This has since been implemented and is available in current builds. I'm not sure what all PSRAM configurations are available, for me it works fine on boards without PSRAM so could be a start.

#12750 open bug for WiFI found on WROVER if anyone is able to confirm on same or other S2 chips.