micropython/micropython #5960: M4/M7 CAN Buffer Message Ordering

kwagyeman, 2020-04-23:
I'm working on an interface library for the OpenMV Cam and I have reliable comms working over I2C/SPI/UART and CAN on the OpenMV Cam H7.

However, CAN will not be usable with the M4/M7. This is because the current code just puts a message into 1 of 3 buffers blindly and those buffers are sent out of order. On the H7 there's a TX FIFO for FDCAN so messages are sent in-order.

I would like to modify the PYB CAN Transmit (can.c) to consider a message transmitted when it is queued in a mailbox versus when the mailbox is actually transmitted and modify the transmit method to try to use the same buffer as a previously queued message with the same id so as to order messages with the same id. The timeout would then be defined on waiting for putting a message in the buffer.

Anyway, I'm asking about how I should go about this. I can just make the changes on the OpenMV Cam fork to fix the library how we need. However, I'd like to do it in a way where it can be up streamed.

Note that CAN_Transmit for fdcan.c works like how I want to change can.c CAN_Transmit now.

If I should create another transmit method that can be done too.

dpgeorge, 2020-04-23:
> Anyway, I'm asking about how I should go about this. I can just make the changes on the OpenMV Cam fork to fix the library how we need. However, I'd like to do it in a way where it can be up streamed.

I'm definitely supportive of having the code here, so thanks for considering that path.

> Note that CAN_Transmit for fdcan.c works like how I want to change can.c CAN_Transmit now.
> 
> If I should create another transmit method that can be done too.

It sounds like the TX behaviour will be quite different to what it currently is.  So the current behaviour should remain, and there are a few options to do that:
1. have a compile-time switch to select the different TX behaviours
1. have a dynamic way to select the behaviour, eg `can.init(tx_fifo=True)`
1. have a new TX method with the new behaviour, so the user can pick either

It sounds like the behaviour is useful, so perhaps 2 or 3 are a good option.  If the signature for the new TX behaviour is the same as the existing then may 2 is the way to go?

hardkrash, 2020-05-08:
Can this not be accomplished with the TXFP bit?

From ST document dm00135183.
> By transmit request order
>
> The transmit mailboxes can be configured as a transmit FIFO by setting the TXFP bit in the CAN_MCR register. In this mode the priority order is given by the transmit request order.
>
> This mode is very useful for segmented transmission.

I considered trying this by setting the bit using the STM32 module after initing the CAN interface.
But I have not had a chance to try this.

kwagyeman, 2020-05-08:
Good catch! That would allow me to easily get this working on the M7. Actually, this bit should probably just be on all the time in general.