micropython/micropython #5963: zephyr-port: Error when Linking C executable zephyr_prebuilt.elf

puchm, 2020-04-23:
Hi,
I am trying around with micropython for zephyr and I got this error when running `make` (without any board specified):

``[ 91%] Linking C executable zephyr_prebuilt.elf
Memory region         Used Size  Region Size  %age Used
             RAM:      387952 B         4 MB      9.25%
        IDT_LIST:         128 B         2 KB      6.25%
/home/moritz/zephyr-sdk-0.11.2/x86_64-zephyr-elf/bin/../lib/gcc/x86_64-zephyr-elf/9.2.0/../../../../x86_64-zephyr-elf/bin/ld: /home/moritz/zephyrproject/micropython/ports/zephyr/libmicropython.a(modzephyr.o): in function ` thread_stack_dump':
/home/moritz/zephyrproject/micropython/ports/zephyr/modzephyr.c:61: undefined reference to `stack_analyze'
collect2: error: ld returned 1 exit status
zephyr/CMakeFiles/zephyr_prebuilt.dir/build.make:121: recipe for target 'zephyr/zephyr_prebuilt.elf' failed
make[3]: *** [zephyr/zephyr_prebuilt.elf] Error 1
make[3]: Leaving directory '/home/moritz/zephyrproject/micropython/ports/zephyr/outdir/qemu_x86'
CMakeFiles/Makefile2:2271: recipe for target 'zephyr/CMakeFiles/zephyr_prebuilt.dir/all' failed
make[2]: *** [zephyr/CMakeFiles/zephyr_prebuilt.dir/all] Error 2
make[2]: Leaving directory '/home/moritz/zephyrproject/micropython/ports/zephyr/outdir/qemu_x86'
Makefile:103: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/home/moritz/zephyrproject/micropython/ports/zephyr/outdir/qemu_x86'
Makefile:86: recipe for target 'all' failed
make: *** [all] Error 2
``

I did the setup of Zephyr following the documentation found [here](https://docs.zephyrproject.org/latest/getting_started/index.html) and then ran `make`. The development toolchain as well as the environment variables are also set up following the same documentation page.
What is the solution for this?
Thank you for your help.

-Moritz

dpgeorge, 2020-04-23:
Which version of zephyr do you use?  Using zephyr 2.2.0 and SDK 0.11.1 it builds ok for me (although does give a warning about deprecated `stack_analyze`).

puchm, 2020-04-23:
I am not quite sure how to find out which version I am using so I got you quite a few versions:

- `west --version` gives me `v0.7.2`
- SDK version is `0.11.2`
- the `VERSION` file in my `zephyrproject` contains the following:
```
VERSION_MAJOR = 2
VERSION_MINOR = 2
PATCHLEVEL = 99
VERSION_TWEAK = 0
EXTRAVERSION =
```
- Operating system is Ubuntu 18.04.4 LTS
- Something else to note: `python` is aliased to `python3`, since I needed that for another project. I can undo that if it is important.

Let me know if you need anything else.
Thank you for your help!

dpgeorge, 2020-04-24:
If `PATCHLEVEL = 99` I'm guessing that means you have a non-stable (non-tagged) release.

Try using the release from this page: https://github.com/zephyrproject-rtos/zephyr/releases/tag/zephyr-v2.2.0

puchm, 2020-04-24:
Thank you so much for your help - building micropython worked now. I have one more question that is on a different topic, maybe you can help me with that...

What is confusing to me is where I would put my Python code. I mean I have built Micropython for Zephyr but so far I have not used any code of my own. So where would I put my Python source files? And how would I build a Zephyr build with my python code in it?

Thank you for your help so far, I would appreciate it if you could give me some hints to that question.
-Moritz

dpgeorge, 2020-04-27:
> What is confusing to me is where I would put my Python code.

This question is best suited to https://forum.micropython.org.  In short: you can either freeze your scripts by putting them in to the `ports/zephyr/scripts/` directory and building, or by creating a filesystem on the board and storing them there.

puchm, 2020-04-27:
Thank you so much! I figured it out.

closed 2020-04-27 by puchm

Carlos-Augusto-Ribeiro, 2020-05-04:
Hi @puchm ,
I have this same problem and I have been trying to solve it for 1 week and I can't. What exactly did you do to get it resolved?

puchm, 2020-05-04:
Hi,
As mentioned above by @dpgeorge , you need to use version `2.2.0`. You do this using some extra parameters when creating the Zephyr project. So you have to redo the installation from the point where you do `west init zephyrproject`.

Instead you use the `-m` and `--mr` flags to specify a specific version like so:
`west init zephyrproject -m https://github.com/zephyrproject-rtos/zephyr --mr v2.2.0`
This is also documented here: https://docs.zephyrproject.org/latest/guides/west/repo-tool.html

Hope that helps!
-Moritz

MaureenHelm, 2020-05-04:
> As mentioned above by @dpgeorge , you need to use version `2.2.0`. You do this using some extra parameters when creating the Zephyr project. So you have to redo the installation from the point where you do `west init zephyrproject`.
> 
> Instead you use the `-m` and `--mr` flags to specify a specific version like so:
> `west init zephyrproject -m https://github.com/zephyrproject-rtos/zephyr --mr v2.2.0`
> This is also documented here: [docs.zephyrproject.org/latest/guides/west/repo-tool.html](https://docs.zephyrproject.org/latest/guides/west/repo-tool.html)

If you want to use the latest zephyr master branch, please see the fix in #6005

You don't have to redo the zephyr installation if you want to switch from master to a tagged release. You can do this instead:
```sh
$ cd zephyrproject/zephyr
$ git checkout v2.2.0
$ west update
```

Carlos-Augusto-Ribeiro, 2020-05-04:
Thank you so much @puchm and @dpgeorge 

I solved the problem and it is now working perfectly.
But in my case I also had to install SDK 0.11.1

It's awesome Python running on the board

dpgeorge, 2020-05-05:
Might be worth adding this info about setting up the build env to the `ports/zephyr/README.md` file.

dpgeorge, 2020-08-28:
Notes about the required Zephyr version and how to install/update the SDK were added in a93a378e933abc294de6ae26fb86a421e858fbcf