micropython/micropython #5970: ESP32: Building firmware failed

robert-hh, 2020-04-25:
Today I tried to re-build the ESP32 firmware and it fails for SPIRAM devices. I re-cloned both ESP-IDF and the Micropython repository with the same result. Log below:
```
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5016
load:0x40078000,len:11132
load:0x40080400,len:5776
entry 0x400806bc
I (650) psram: This chip is ESP32-D0WD
I (650) spiram: Found 32MBit SPI RAM device
I (651) spiram: SPI RAM mode: flash 40m sram 40m
I (653) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (661) cpu_start: Pro cpu up.
I (664) cpu_start: Application information:
I (669) cpu_start: Compile time:     Apr 25 2020 15:40:16
I (675) cpu_start: ELF file SHA256:  0000000000000000...
I (681) cpu_start: ESP-IDF:          v3.3.2
I (686) cpu_start: Starting app cpu, entry point is 0x4008386c
I (678) cpu_start: App cpu up.
I (1572) spiram: SPI SRAM memory test OK
I (1572) heap_init: Initializing. RAM available for dynamic allocation:
I (1572) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (1578) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (1585) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (1591) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (1597) heap_init: At 3FFCAB80 len 00015480 (85 KiB): DRAM
I (1603) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1610) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1616) heap_init: At 40099BBC len 00006444 (25 KiB): IRAM
I (1622) cpu_start: Pro cpu start user code
I (299) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x40097d24  PS      : 0x00060434  A0      : 0x80098c93  A1      : 0x3ffbae40  
A2      : 0x3ffbdf64  A3      : 0x00060423  A4      : 0x00000001  A5      : 0x0000abab  
A6      : 0xb33fffff  A7      : 0x3ffb63ec  A8      : 0x0000abab  A9      : 0x3ffbae30  
A10     : 0x0000cdcd  A11     : 0x00000780  A12     : 0x3ffbd31c  A13     : 0x000004e0  
A14     : 0x3ffb9a30  A15     : 0x3ffbd619  SAR     : 0x00000011  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x400942fd  LEND    : 0x40094301  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

Backtrace: 0x40097d24:0x3ffbae40 0x40098c90:0x3ffbae70 0x40096638:0x3ffbae90 0x400969dc:0x3ffbaeb0 0x40096ab9:0x3ffbaef0 0x40096ad1:0x3ffbaf10 0x40086c9e:0x3ffbaf30 0x40086cc7:0x3ffbaf50 0x40086e25:0x3ffbaf90 0x4017ed4d:0x3ffbafb0 0x40179fe5:0x3ffbb270 0x40179f6d:0x3ffbb2c0 0x4009707d:0x3ffbb2f0 0x400971e0:0x3ffbb310 0x40097b3c:0x3ffbb330 0x40099710:0x3ffbb350 0x400996c6:0x3ffb79b8

Core 1 register dump:
PC      : 0x40097d1c  PS      : 0x00060934  A0      : 0x80086c88  A1      : 0x3ffb98b0  
A2      : 0x3ffbf124  A3      : 0x00060723  A4      : 0x00000001  A5      : 0x0000cdcd  
A6      : 0xb33fffff  A7      : 0x3ffb9e98  A8      : 0x0000cdcd  A9      : 0x3ffb9ca0  
A10     : 0x0000abab  A11     : 0x00060023  A12     : 0x00000001  A13     : 0x0000abab  
A14     : 0x00000000  A15     : 0x00000001  SAR     : 0x00000018  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

Backtrace: 0x40097d1c:0x3ffb98b0 0x40086c85:0x3ffb98e0 0x40086cc7:0x3ffb9900 0x40086e25:0x3ffb9940 0x4017ed4d:0x3ffb9960 0x40179fe5:0x3ffb9c20 0x40179f6d:0x3ffb9c70 0x40097361:0x3ffb9ca0 0x40097a53:0x3ffb9cd0 0x40099710:0x3ffb9cf0 0x400996c6:0x00000001

```
Both  PC addresses are somewhere in task.c. The non-SPIRAM version works. The board itself is fine. It can run the the pre-built images. The compiler is xtensa-esp32-elf-linux64-1.22.0-98-g4638c4f-5.2.0. What changed in the last week or so?

tve, 2020-04-25:
I just built the TINYPICO version from master and it works. I also built GENERIC_SPIRAM and it starts OK on a tinypico board. I don't have a wrover handy.
You're not saying what type of board/module you're using, nor which git commit you're building at, nor which board defs you're building...
Have you tried http://micropython.org/download/esp32/ ?
Have you decoded the backtrace, e.g. using https://github.com/tve/esp32-backtrace ?

robert-hh, 2020-04-25:
It is a Wemos LOLIN32 Pro. The build works fine for a non-SPIRAM board, and like three days ago I also made build to a board with e ESP32 PICO, but w/o SPIRAM.
The prebuilt packages work too. And I use the same procedure as always, so I'm a little bit puzzled. That's what backtrace tells:
```
BT-11: 0x40176c49 is in _vfiprintf_r (../../../.././newlib/libc/stdio/vfprintf.c:860).
860	../../../.././newlib/libc/stdio/vfprintf.c: Datei oder Verzeichnis nicht gefunden.
BT-12: 0x40171ee5 is in fiprintf (../../../.././newlib/libc/stdio/fiprintf.c:50).
50	../../../.././newlib/libc/stdio/fiprintf.c: Datei oder Verzeichnis nicht gefunden.
BT-13: 0x40171e6d is in __assert_func (../../../.././newlib/libc/stdlib/assert.c:59).
59	../../../.././newlib/libc/stdlib/assert.c: Datei oder Verzeichnis nicht gefunden.
```
These files newlib/libc/stdio/vprintf.c etc. doe not exist in the SDK

tve, 2020-04-25:
If you set ESPIDF to an absolute path you avoid those crazy ../.../../.. and you make sure you're not picking up something odd.
You can also set TOP as an abs path to the top dir of MP.
Your backtrace starts at frame 11, which is odd.

robert-hh, 2020-04-25:
Setting ESPIDF does not change the log Setting TOP . I get also backtraces starting at frame 4, but with similar content. The mentioned files do not exist in the esp-idf.
```
Core 0:
BT-4: 0x4017ed4d is in _vfiprintf_r (../../../.././newlib/libc/stdio/vfprintf.c:860).
860	../../../.././newlib/libc/stdio/vfprintf.c: Datei oder Verzeichnis nicht gefunden.
BT-5: 0x40179fe5 is in fiprintf (../../../.././newlib/libc/stdio/fiprintf.c:50).
50	../../../.././newlib/libc/stdio/fiprintf.c: Datei oder Verzeichnis nicht gefunden.
BT-6: 0x40179f6d is in __assert_func (../../../.././newlib/libc/stdlib/assert.c:59).
59	../../../.././newlib/libc/stdlib/assert.c: Datei oder Verzeichnis nicht gefunden.

Core 1:
BT-9: 0x4017ed4d is in _vfiprintf_r (../../../.././newlib/libc/stdio/vfprintf.c:860).
860	../../../.././newlib/libc/stdio/vfprintf.c: Datei oder Verzeichnis nicht gefunden.
BT-10: 0x40179fe5 is in fiprintf (../../../.././newlib/libc/stdio/fiprintf.c:50).
50	../../../.././newlib/libc/stdio/fiprintf.c: Datei oder Verzeichnis nicht gefunden.
BT-11: 0x40179f6d is in __assert_func (../../../.././newlib/libc/stdlib/assert.c:59).
59	../../../.././newlib/libc/stdlib/assert.c: Datei oder Verzeichnis nicht gefunden.
```

robert-hh, 2020-04-25:
I tried it with another SPIRAM board,  a Pycom LoPy4. Same result. So I assume it's not the board, but something in my development environment. Which compiler version are you using?

robert-hh, 2020-04-25:
Suddenly it worked again. That is also not really satisfying. But thanks a lot for the backtrace tool. I had seen that in the discussions considered it as something sometimes useful. But at this very  moment I forgot it.

closed 2020-04-25 by robert-hh

jimmo, 2020-04-25:
> The compiler is xtensa-esp32-elf-linux64-1.22.0-98-g4638c4f-5.2.0. What changed in the last week or so?

FYI that compiler is very old. I don't think this explains your issue necessarily, but there's a much more recent gcc version available but running the `install.sh` script in the IDF. (Although this is only in IDF 4). More info in esp32/README.md

tve, 2020-04-26:
if you don't want esp-idf to install the whole world, the easiest way to install stuff is to follow the commands in `.travis.yml`...

jimmo, 2020-04-26:
> if you don't want esp-idf to install the whole world, the easiest way to install stuff is to follow the commands in `.travis.yml`...

Yes but be aware that will also get you gcc 5.2. Like I said, I don't think this is the reason for Robert's problem (and the fact that the travis build and (afaik) the official builds use gcc 5.2 should confirm that).

tve, 2020-04-26:
> Yes but be aware that will also get you gcc 5.2. Like I said, I don't think this is the reason for Robert's problem (and the fact that the travis build and (afaik) the official builds use gcc 5.2 [...]

Quick note: I submitted a now-merged PR a few days ago that separates the esp-idf v3 and v4 builds in travis and changed the compiler in the process: https://github.com/micropython/micropython/blob/master/.travis.yml#L264

robert-hh, 2020-04-26:
I have two variants of the IDF installed. xtensa-esp32-elf-gcc --version
xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-98-g4638c4f) 5.2.0
and
xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
The first one was downloaded from the espressif web site, the second one created with crosstool. As far as I understand it, both are gcc Version 5.2.0. Both seem to work equally with other firmware stacks. I have still no clue why the problem occurred and went away.
Edit: I installed the version pointed at in .travis.yml, and creating the firmware with that gcc 8.2.0 works too, and it seems that the created image is slightly smaller. ~146k vs. ~147k for the app.