micropython/micropython #5971: zephyr port: boards?

tve, 2020-04-26:
The zephyr port evidently does not have a board subdirectory. Are the proj_*.conf files the list of supported or tested boards? Nothing is mentioned in the readme. Also, the contents of those files doesn't provide much of a clue about the boards. Fortunately a search for the cryptic codes in the file names brings up reasonable results, but it sure would have been nice to have a comment there spelling out the full name of the board an maybe even a link...

dpgeorge, 2020-04-27:
> Are the proj_*.conf files the list of supported or tested boards? Nothing is mentioned in the readme.

The README states:
```
This port requires Zephyr version 1.8 or higher. All boards supported
by Zephyr (with standard level of features support, like UART console)
should work with MicroPython (but not all were tested).

...

If you don't specify BOARD, the default is `qemu_x86` (x86 target running
in QEMU emulator). Consult Zephyr documentation above for the list of
supported boards.
```

Since Zephyr supports a very large number of boards it's not practical to document them in this port, and keep the list maintained.  Instead, it's understood that users of this port would first have some familiarity with Zephyr.  For example how pins and peripherals are named.

tve, 2020-04-27:
I didn't ask for a list of boards that can be made to work, which indeed would be unmaintainable. I was looking a the directory and wondering what it would take for me to try it out. I guess it's not worth typing a paragraph of text to let someone like me know which values of `make BOARD=...` are supported  by the existing configuration files...

closed 2020-04-27 by tve

dpgeorge, 2020-04-28:
> I was looking a the directory and wondering what it would take for me to try it out.

I think the README is pretty OK in describing how to try it out, especially with qemu.

>  which values of `make BOARD=...` are supported by the existing configuration files

All boards in Zephyr are supported by the existing configuration files (they'll use `prj_base.conf` as their base configuration), so if you have such a board then `make BOARD=<zephyr-board-name>` should work.  Additional configuration is given for some boards, and these can be used to show how to configure your particular board for more board-specific features.