micropython/micropython #5974: armcc:Why do I get an error when compiling nrl_push in the nrljump.c file using the armcc compiler.

xiaoxiaoj, 2020-04-26:
Why do I get an error when compiling nrl_push in the nrljump.c file using the armcc compiler.
the error is as follows:
line 43: Error:  #549: variable "R0" is used before its value is set
"str r4,[r0,#12] \n"
line 43: Error:  #549: variable "R4" is used before its value is set
"str r4,[r0,#12] \n"

How do I need to solve this problem.

dpgeorge, 2020-04-26:
It's most likely because of an old compiler version, or that armcc doesn't work well with the attributes and asm statement in that file. Try using arm-none-eabi-gcc.

xiaoxiaoj, 2020-04-28:
This is because the version of armcc compiler I used is too old.The attribute is supported on Arm compiler 6 which is armclang compiler.The arm-none-eabi-gcc is ok.thank you very mach.

closed 2020-04-28 by dpgeorge