micropython/micropython #5980: py: starred expression incompatibility with cpython

tve, 2020-04-28:
I don't know whether this is CPython being lenient or MicroPython not accepting something valid:
```
$ cat star.py
import struct
buf = bytearray(10)
print("ints: %d %d %d" % (*struct.unpack_from("!II", buf, 2), 55))
$ python star.py
ints: 0 0 55
$ pyboard star.py
Traceback (most recent call last):
  File "<stdin>", line 3, in <module>
SyntaxError: *x must be assignment target
```

peterhinch, 2020-04-28:
This can be reduced to
```python
a = (0, 0)
(*a, 55)
```
which produces (0, 0, 55) under CPython but errors under uP.

See [PEP448](https://www.python.org/dev/peps/pep-0448/) which hasn't yet been implemented in uP.

tve, 2020-04-28:
Thanks Peter! All the Python versions are confusing... In any case, this issue is tracked in #1329 so no need to keep this one open.

closed 2020-04-28 by tve