micropython/micropython #5981: uos.stat not works: esp32 VfsLfs2

straga, 2020-04-28:
```python
>>> uos.listdir("./")
['_conf', 'core', 'lib', 'main.py', 'mod', 'u_config']

>>> uos.stat("./")
(16384, 0, 0, 0, 0, 0, 0, 0, 0, 0)

>>> uos.stat("./main.py")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 2] ENOENT

>>> uos.stat("./core")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 2] ENOENT
>>>
```

peterhinch, 2020-04-28:
Did you erase flash before installing the firmware?

I erased flash and installed today's build. The above worked without error.

straga, 2020-04-28:
Yep. Totaly.
More info.
Create folder factory
uos.chdir("/{}".format("factory"))
And try now.

robert-hh, 2020-04-28:
The sequence above works well. And no main.py and boot.py always have to be in the topmost folder. 
Besides that, littlefs seem not to do filename normalization. If you execute_
uos.chdir("/factory")
uos.chdir("..")
uos.getcwd()
I would expect "/" as result, but I get "/factory/.."
And, having ~70 files in my file system, littlfs is awfully slow in directory operations.

peterhinch, 2020-04-28:
@straga I cannot replicate the `ENOENT` exception with any of your samples.

@robert-hh I can confirm your observation re `uos.chdir('..')`: it works on a FAT SD card but produces that strange result on the ESP32 littlefs partition. Yet `uos.chdir('/')` works. This is surely a bug.

I tested my Flash and EEPROM drivers extensively with 128 files under littlefs, with external chips. My aim was checking for corruption with changing file sizes, file deletion and creation. I didn't measure the speed of directory operations - they seemed fast so I saw no need. But my testing was with Pyboards and ESP8266. I wonder if this is an ESP32 issue?

robert-hh, 2020-04-28:
Yes. That was on.a ESP32. Never seen that with FAT.

robert-hh, 2020-04-28:
There is more to fail. example:
```
from uos import *
mkdir("test")
chdir("test")
getcwd()  # returns "/test"
chdir(".") 
getcwd() # returns "/test/."
chdir ("..")
getcwd() #  returns "/test./.."
chdir("test") 
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: -2
```
... and so on. The Pycom branch for instance works well with that sequence. They have dedicated checks for '.' and '..' in chdir, so they won't get confused.

tve, 2020-04-28:
Have you guys looked at the LittleFS docs to see whether this is "as designed"?

peterhinch, 2020-04-29:
The LittleFS [design doc](https://github.com/ARMmbed/littlefs/blob/master/DESIGN.md) and [spec](https://github.com/ARMmbed/littlefs/blob/master/SPEC.md) are low-level definitions and don't define the OS interface.

robert-hh, 2020-04-29:
No, but even if, it's a pain ..... and can be fixed.

tve, 2020-04-29:
haha, LittleFS doesn't have a notion of CWD, that's all in MP code. Pull request?

robert-hh, 2020-04-29:
Yes. That should be fixed. I have spotted already yesterday eve the place to do so. That's in vlfsx.c around line 276 or 284. And code to do path name normalization should exists. Maybe in the FAT section.  Even I have written something like that in the past, and recently in Python.

straga, 2020-04-29:
More Info

``` python
>>> uos.stat("/")
(16384, 0, 0, 0, 0, 0, 0, 0, 0, 0)

>>> uos.stat("./")
(16384, 0, 0, 0, 0, 0, 1, 0, 0, 0)

>>> uos.stat(".")
(16384, 0, 0, 0, 0, 0, 1, 0, 0, 0)

>>> uos.listdir("/")
['boot.py', 'factory']

>>> uos.stat("/boot.py")
(32768, 0, 0, 0, 0, 0, 139, 0, 0, 0)

>>> uos.stat("/factory")
(16384, 0, 0, 0, 0, 0, 1, 0, 0, 0)

>>> uos.listdir("./")
['boot.py', 'factory']

>>> uos.stat("./boot.py")
(32768, 0, 0, 0, 0, 0, 139, 0, 0, 0)

>>> uos.stat("./factory")
(16384, 0, 0, 0, 0, 0, 1, 0, 0, 0)

>>> uos.listdir(".")
['boot.py', 'factory']

>>> uos.stat("boot.py")
(32768, 0, 0, 0, 0, 0, 139, 0, 0, 0)

>>> uos.stat("factory")
(16384, 0, 0, 0, 0, 0, 1, 0, 0, 0)

import sys
part_name = "factory"
sys.path.append("/{}".format(part_name))
uos.chdir("/{}".format(part_name))

>>> uos.stat("/")
(16384, 0, 0, 0, 0, 0, 0, 0, 0, 0)

>>> uos.stat("./")
(16384, 0, 0, 0, 0, 0, 1, 0, 0, 0)

>>> uos.stat(".")
(16384, 0, 0, 0, 0, 0, 1, 0, 0, 0)

>>> uos.listdir("/")
['boot.py', 'factory']

>>> uos.stat("/boot.py")
(32768, 0, 0, 0, 0, 0, 139, 0, 0, 0)

>>> uos.stat("/factory")
(16384, 0, 0, 0, 0, 0, 1, 0, 0, 0)

>>> uos.listdir("./")
['main.py']

>>> uos.stat("./main.py")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 2] ENOENT
>>>

>>> uos.listdir(".")
['main.py']
>>> uos.stat("main.py")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 2] ENOENT
>>>

>>> uos.stat("/factory/main.py")
(32768, 0, 0, 0, 0, 0, 2736, 0, 0, 0)
>>>
```

robert-hh, 2020-04-29:
OK. I have a piece of code that, as far as I could tell, deals with the .. and . and // in the path names. It is not small, and it is not elegant. I have two questions about it:
- I use m_malloc() to allocate space for the new path name from the heap. How do I prevent gc_collect() from stealing it, before it is used. Not likely, but I did not follow all possible cases.
- How do I call our nice code formatter on that?
Code below. It sits on top of vfs_lfsx.c.
```
// canonalize path, remove ., .. and // items
static char *trim_path(const char *path) {
    char *token[32];  // worst size is MAX_PATH / 2
    char *new_path = m_malloc(strlen(path) + 1);
    const size_t items = sizeof(token) / sizeof(char *);
    char *in;
    bool absolute = (path[0] == '/');
    size_t n = 0, i;

    // Break path into pieces. 
    strcpy(new_path, path);
    in = new_path;
    for (n = 0; n < items; n++) {
        token[n] = strtok(in, "/");
        if (token[n] == NULL) {
            break;
        }
        in = NULL;
    }
    // scan for . and .. and drop them
    for (i = 0; i < n; i++) {
        if (strcmp(token[i], ".") == 0) { // drop the  '.' item
            for (size_t j = i; j < (items - 1); j++) {
                token[j] = token[j + 1];
            }
            i -= 1;
            n -= 1;
        } else if (strcmp(token[i], "..") == 0) { // drop the '..' and the previous item
            if (i >= 1) { // drop the '..' and the previous item
                for (size_t j = i - 1; j < (items - 2); j++) {
                    token[j] = token[j + 2];
                }
                i -= 2;
                n -= 2;
            } else { // At the top, just drop the  '..' item
                for (size_t j = i; j < (items - 1); j++) {
                    token[j] = token[j + 1];
                }
                i -= 1;
                n -= 1;
            } 
        }
    }
    // Rebuild the path
    size_t p_size = 0;
    if (absolute) {
        new_path[0] = '/';
        p_size = 1;
    }
    if (n != 0) {  // nothing left
        for (i = 0; i < n; i++) {
            if (i > 0) {
                new_path[p_size++] = '/';
            }
            memcpy(new_path + p_size, token[i], strlen(token[i]));
            p_size += strlen(token[i]);
        }
    }
    new_path[p_size] = '\0';
    return new_path;
}

const char *MP_VFS_LFSx(make_path)(MP_OBJ_VFS_LFSx * self, mp_obj_t path_in) {
    const char *path = mp_obj_str_get_str(path_in);
    if (path[0] != '/') {
        size_t l = vstr_len(&self->cur_dir);
        if (l > 0) {
            vstr_add_str(&self->cur_dir, path);
            path = vstr_null_terminated_str(&self->cur_dir);
            self->cur_dir.len = l;
        }
    }
    return trim_path(path);
}
```

tve, 2020-04-29:
Time to drop this into a PR so it can be reviewed ;-)

robert-hh, 2020-04-29:
OK. PR is done.

dpgeorge, 2020-05-08:
The stat issue was fixed in a5ea4b9f3f3fd394afdf65223b5e55c8d9fccc53, chdir normalisation in d3ea28d04a7df9ca536a9002c8fda2f6e3a88f09 and rename fix in 0f83ef395cccaa771543450ff8a4200da108e3d6

closed 2020-05-08 by dpgeorge