micropython/micropython #5983: 'closure' object has no attribute '__name__'

pumelo, 2020-04-28:
```
def dummy():
    x = True

    def func1():
        pass

    def func2():
        nonlocal x
        pass
    print(func1.__name__)
    print(func2.__name__)

dummy()
```

expected output (test with cpython):
```
func1
func2
```

actual output with 1.12.0 on esp32:
```
func1
Traceback (most recent call last):
  File "<stdin>", line 13, in <module>
  File "<stdin>", line 11, in dummy
AttributeError: 'closure' object has no attribute '__name__'
```

dpgeorge, 2020-05-03:
This is not implemented, mainly because it increases the size of the code and it's a feature that's not used very often.  What is the use-case for this, can it be done some other way?

pumelo, 2020-05-03:
😄  There are always workarounds. The use case was for a debug function which should have printed the name of the function.

mikebentley15, 2021-10-03:
May I open up this issue again for discussion?  I find not having the `__name__` attribute for closures makes writing and porting existing function decorators challenging.

I actually prototyped up a change and can issue a pull request.  My implementation simply implements an `attr` function and forwards it on to its `self->fun` object.  This allows getting any attribute that is defined in the `function` class also from the `closure` class.

```
diff --git a/py/objclosure.c b/py/objclosure.c
index 054b65789..bdad6793f 100644
--- a/py/objclosure.c
+++ b/py/objclosure.c
@@ -78,6 +78,18 @@ STATIC void closure_print(const mp_print_t *print, mp_obj_t o_in, mp_print_kind_
 }
 #endif
 
+#if MICROPY_PY_FUNCTION_ATTRS
+void mp_obj_closure_attr(mp_obj_t self_in, qstr attr, mp_obj_t *dest) {
+    if (dest[0] != MP_OBJ_NULL) {
+        // not load attribute
+        return;
+    }
+    // else, forward to self_in->fun
+    mp_obj_closure_t *o = MP_OBJ_TO_PTR(self_in);
+    mp_load_method_maybe(o->fun, attr, dest);
+}
+#endif
+
 const mp_obj_type_t mp_type_closure = {
     { &mp_type_type },
     .flags = MP_TYPE_FLAG_BINDS_SELF,
@@ -86,6 +98,9 @@ const mp_obj_type_t mp_type_closure = {
     .print = closure_print,
     #endif
     .call = closure_call,
+    #if MICROPY_PY_FUNCTION_ATTRS
+    .attr = mp_obj_closure_attr,
+    #endif
 };
 
 mp_obj_t mp_obj_new_closure(mp_obj_t fun, size_t n_closed_over, const mp_obj_t *closed) {
```

Here's an example of before this change:

```
MicroPython ef52f0d-dirty on 2021-09-14; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> def f():
...     a = 1
...     def g():
...         print(a)
...     return g
... 
>>> h = f()
>>> h
<closure <function g at 0x7f8683352940> at 7f8683352980, n_closed=1 <cell 3 1> >
>>> dir(h)
['__class__']
>>> dir(f)
['__class__', '__name__', '__globals__']
>>> h.__class__
<class 'closure'>
>>> h.__name__
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'closure' object has no attribute '__name__'
>>> h.__globals__
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'closure' object has no attribute '__globals__'
>>> h.doesnotexist
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'closure' object has no attribute 'doesnotexist'
```

and after this change

```
MicroPython v1.17-68-g35fb90bd5-dirty on 2021-10-02; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> def f():
...     a = 1
...     def g():
...         print(a)
...     return g
... 
>>> h = f()
>>> h
<closure <function g at 0x7f4a4681c940> at 7f4a4681c980, n_closed=1 <cell 3 1> >
>>> dir(h)
['__class__', '__name__', '__globals__']
>>> dir(f)
['__class__', '__name__', '__globals__']
>>> h.__class__
<class 'closure'>
>>> h.__name__
'g'
>>> h.__globals__
{'h': <closure <function g at 0x7f4a4681c940> at 7f4a4681c980, n_closed=1 <cell 3 1> >, '__name__': '__main__', 'f': <function f at 0x7f4a4681c8c0>}
>>> h.doesnotexist
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'closure' object has no attribute 'doesnotexist'
```

dpgeorge, 2022-06-24:
Implemented by d68532558d0f60737ccd5a38ae33f719333520a4

closed 2022-06-24 by dpgeorge

hustyangyi, 2023-11-02:
=== def AsyncCheckNone(func):
===     async def wrap(*params):
===         if any(x is None for x in params):
===             print(f"{func.__name__}: invalid param")
===         else:
===             result = await func(*params)
===             return result
===     print(f"wrap: {dir(wrap)}")
===     print(f"{func.__name__}")
===     print(f"{wrap.__name__}")
===     wrap.__name__ = func.__name__
===     return wrap
===
>>> AsyncCheckNone(afunc)
wrap: ['__class__', '__name__', '__globals__']
afunc
wrap
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<stdin>", line 11, in AsyncCheckNone
AttributeError: 'closure' object has no attribute '__name__'

Cannot set the __name__ attribute of a closure

jimmo, 2023-11-02:
> Cannot set the `__name__` attribute of a closure

This is expected behavior. These attributes are read-only in MicroPython.