micropython/micropython #5984: Seems like 3.3.2 SDK caused more wifi network issues

petrkr, 2020-04-28:
First sorry about 3.3.2 at #5554 

seems like at 3.3.2 they introduced some bug regarding stability. It does not have problems with 'bcn timeouts' anymore. But it will stop communication  after while... Regards testing seems every hours or so...

At 3.3.1 (with memory hack to support other AP than from UPC) seems to be fine.

See

https://parallelgarden.surikata.info/d/aZO-SFaZz/octopuslab-upy?orgId=1&refresh=30s

Graphs "influx demo".

Do not know exactly what to report to ESP-IDF team. Because that would be more for them than uPy team.

tve, 2020-04-29:
Have you tried 4.0? It seems unlikely to me that Espressif will spend time fixing 3.x at this point...

petrkr, 2020-04-29:
4.0 causing bcn timeout same as 3.3.1 (tested before when testing 4.0 in considering 3.3.2). Maybe I can try 4.0 with my 'memory hack' Good point.

petrkr, 2020-04-29:
Tested now with 4.0 so I remember well it does same bcn timeout error  as 3.3.1 Let;s try memory hack
```
I (533) cpu_start: Pro cpu up.
I (533) cpu_start: Application information:
I (533) cpu_start: Compile time:     Apr 29 2020 13:49:17
I (537) cpu_start: ELF file SHA256:  0000000000000000...
I (543) cpu_start: ESP-IDF:          v4.0
I (547) cpu_start: Starting app cpu, entry point is 0x40082818
I (539) cpu_start: App cpu up.
I (558) heap_init: Initializing. RAM available for dynamic allocation:
I (565) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (571) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (577) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (583) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (589) heap_init: At 3FFCCA48 len 000135B8 (77 KiB): DRAM
I (595) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (602) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (608) heap_init: At 4009E954 len 000016AC (5 KiB): IRAM
I (614) cpu_start: Pro cpu start user code
I (633) spi_flash: detected chip: generic
I (633) spi_flash: flash io: dio
I (634) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
MicroPython v1.12-401-g7b5b109ff-build-octopusLAB on 2020-04-29; ESP32 module with ESP32

I (32652) modsocket: Initializing
Installing shutil
Installing to: /lib/
I (40742) wifi: bcn_timout,ap_probe_send_start
I (43242) wifi: ap_probe_send over, resett wifi status to disassoc
I (43242) wifi: state: run -> init (c800)
I (43242) wifi: pm stop, total sleep time: 5526634 us / 15777115 us

I (43242) wifi: new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
mbedtls_ssl_handshake error: -71
```

petrkr, 2020-04-29:
Yes with -2048 memory "hack" that 4.0 seems to work (atleast with this AP, some other AP needs -4096 and with BLE even -8kB)

```
I (533) cpu_start: Pro cpu up.
I (533) cpu_start: Application information:
I (533) cpu_start: Compile time:     Apr 29 2020 13:49:17
I (537) cpu_start: ELF file SHA256:  0000000000000000...
I (543) cpu_start: ESP-IDF:          v4.0
I (547) cpu_start: Starting app cpu, entry point is 0x40082818
I (539) cpu_start: App cpu up.
I (558) heap_init: Initializing. RAM available for dynamic allocation:
I (565) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (571) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (577) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (583) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (589) heap_init: At 3FFCCA48 len 000135B8 (77 KiB): DRAM
I (595) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (602) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (608) heap_init: At 4009E954 len 000016AC (5 KiB): IRAM
I (614) cpu_start: Pro cpu start user code
I (633) spi_flash: detected chip: generic
I (633) spi_flash: flash io: dio
I (634) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
MicroPython v1.12-401-g7b5b109ff-dirty-build-octopusLAB on 2020-04-29; ESP32 module with ESP32
I (25322) modsocket: Initializing
Installing shutil                                                                                                                                                       
Installing to: /lib/                                                                                                                                                    
Warning: micropython.org SSL certificate is not validated                                                                                                               
Installing micropython-shutil 0.0.3 from https://micropython.org/pi/shutil/shutil-0.0.3.tar.gz                                                                          
Running deploy                                                                                                                                                          
Extracting dir: assets/                                                                                                                                                 
Extracting file: assets/octopus_image.pbm                                                                                                                               
Extracting file: assets/icons9x9.mpy                                                                                                                                    
...
...
```

diff
```
[petrkr@pkdell esp32 (octopus-4.0 *%)]$ git diff
diff --git a/ports/esp32/main.c b/ports/esp32/main.c
index 81e4a6434..f08577244 100644
--- a/ports/esp32/main.c
+++ b/ports/esp32/main.c
@@ -100,7 +100,7 @@ void mp_task(void *pvParameter) {
     }
     #else
     // Allocate the uPy heap using malloc and get the largest available region
-    size_t mp_task_heap_size = heap_caps_get_largest_free_block(MALLOC_CAP_8BIT);
+    size_t mp_task_heap_size = heap_caps_get_largest_free_block(MALLOC_CAP_8BIT)-2048;
     void *mp_task_heap = malloc(mp_task_heap_size);
     #endif```

tve, 2020-04-29:
In your log I see:
```
Installing shutil
Installing to: /lib/
```
what is that? It's not std... Are you perhaps calling upip and this is the same as https://github.com/micropython/micropython/issues/5543#issuecomment-583848383, i.e. out of memory?

petrkr, 2020-04-29:
Yes it is out of mem issue, that is why 'memory hack' will help. Here it happening on upip too, but i tried also urrequests.. actually any TLS connection will do that. Maybe most of users will not notice that as many of them not using TLS at all and follow stupid tutorials which still use HTTP or MQTT instead HTTPS (MQTTS respectively).

Your linked comment is exactly same. Actually In 3.3.2 it it ok (until you enable BLE) but new bug with that stop communication after 1 hour.

My guess is it would be something with WPA rekey period. Unfortunately I can not see rekey period at this router, so it's only guess.

petrkr, 2020-04-29:
But still i'm for go with 3.3.2 and not revert 3.3.1, because this 'bug' can be fixed by try/except anf reconnect (sta.disconnect,connect) from Python.  That bcn error need to be 'workarounded' by change in main.c and recompile binary, so for most of users is better reconnect than reconpile way.

And 4.0 act same as 3.3.1, but 4.1 (or 4.2) are still under development.

petrkr, 2021-08-12:
Switched to 4.x sdk

closed 2021-08-12 by petrkr