micropython/micropython #5985: Confusion about STM32WB55 BLE TX power level implementation

Hendrik-K, 2020-04-29:
Hello,

I was playing around with BLE on my STM32WB55 within MicroPython when I wondered how to change the BLE TX power.
Then I found out that it is set within stm32/rfcore.c:

https://github.com/micropython/micropython/blob/2e3c42775a402749e790474643cb9f776561264d/ports/stm32/rfcore.c#L419

I suppose the two hex escape sequences _"\x00\x06"_ represent the parameters _En_High_Power_ and _PA_Level_ specified within ST's WB55 interface application note in chapter 2.3.4 for the _ACI_HAL_SET_TX_POWER_ command.

The application note specifies that PA_Level should be 0x06 for a -15.25 dBm TX power level and 0x19 for a 0 dBm TX power level.

The application note can be found here: [https://www.st.com/content/ccc/resource/technical/document/application_note/group1/25/56/53/7b/a0/17/47/c8/DM00571230/files/DM00571230.pdf/jcr:content/translations/en.DM00571230.pdf](url)

This left me confused, as the comment at the end of the line doesn't seem to match the operation performed.

Shouldn't it be

`tl_ble_hci_cmd_resp(HCI_OPCODE(OGF_VENDOR, OCF_SET_TX_POWER), 2, (const uint8_t *)"\x00\x19"); // 0 dBm`

if you wanted to set the TX power level to 0 dBm or

`tl_ble_hci_cmd_resp(HCI_OPCODE(OGF_VENDOR, OCF_SET_TX_POWER), 2, (const uint8_t *)"\x00\x06"); // -15.25 dBm`

with the comment changed to the actual TX power level to avoid this confusion?

It may be that I'm just overlooking something and that it is all fine, but that's just my current understanding of the implementation.

dpgeorge, 2020-04-29:
Thanks for the report, it looks like you are right that it doesn't set the TX power level as described in the comment.

Since the default TX power is the maximum of +6dBm it's probably best to just remove that line altogether and let it keep the default at reset.

Then it would be good to allow the user to adjust the value in Python, something like `BLE.config(txpower=...)` (see related #5670).

dpgeorge, 2020-06-16:
Fixed by 4b5dd012e0fe615001bbd1edf2f85ea30371a1d6

closed 2020-06-16 by dpgeorge