micropython/micropython #5990: Can not get any serial console with minicom

nejdetckenobi, 2020-04-29:
First of all, I used ESP8266 and flashed it, used micropython on it. So I have some experience
Here are the steps I tried

1. I erase flash with following command
  
    `esptool.py --chip esp32 --port /dev/ttyUSB0 erase_flash `
  
   The command result does not contain any errors.

2. I write the firmware again with following command

    `esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 115200 write_flash -fm dio -z 0x1000 esp32_workspace/BINFILE`
    
    On `BINFILE` part, I tried the binfiles below
    - esp32-idf3-20191220-v1.12.bin
    - esp32-idf3-20200429-unstable-v1.12-418-g2e3c42775.bin
    - esp32-idf4-20191220-v1.12.bin
    - esp32-idf4-20200422-v1.12-388-g388d419ba.bin

    On every output, verify part worked and detected no errors
    I also used `-fm dio` options (I tried at least 8 times, 2 times per bin file, with and without `-fm dio` option)

3. I disconnect the device and connect it back.
4. I open a minicom session with the command below
    `minicom -b 115200`

I couldn't get any serial console. I think I am doing something wrong but I don't know what it is.

Here are some notes:
- I use linux mint with some recent version kernel so drivers are not an issue.
- I use same cable for my ESP8266 so cable works just fine.

- In case of missing information on the hardware side, this is the image of the device.

[image]

tve, 2020-04-29:
This type of question is better posted on forum.micropython.org

For a raw tty I use `miniterm2.py --filter direct /dev/ttyUSB0 115200` and it's handy that ctrl-t - ctrl-d toggles DTR which allows me to reset the esp32 without poking at the rest button on the board.

You need to make sure that minicom does the right thing with DTR and RTS.

nejdetckenobi, 2020-04-29:
Thanks for replying. 
I am sure my minicom settings are fine. 
I also tried miniterm and picocom but got the same result.
I am following the guide very strictly and following every single advice to get a serial console.

I also tried all methods I previously shared with you on two different esp32s. So I basically eliminated the bad hardware problem. No luck so far. I will also take your advice and create a post on micropython forum. Thanks.

robert-hh, 2020-04-30:
In the minicom call above I miss the name of the USB port. Besides that, Is there any output if you connect to the ESp32 and push the reset/EN button?

nejdetckenobi, 2020-04-30:
Thanks for the reply @robert-hh , I pre configured that. It uses `/dev/ttyUSB0` by default. And I confirm that the device is there

nejdetckenobi, 2020-04-30:
This is the output when I press EN button while I am in minicom

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:4844
load:0x40078000,len:9656
load:0x40080400,len:6216
entry 0x400806f0
I (457) cpu_start: Pro cpu up.
I (458) cpu_start: Application information:
I (458) cpu_start: Compile time:     Dec 20 2019 07:50:41
I (461) cpu_start: ELF file SHA256:  0000000000000000...
I (467) cpu_start: ESP-IDF:          v3.3
I (472) cpu_start: Starting app cpu, entry point is 0x40083600
I (0) cpu_start: App cpu up.
I (482) heap_init: Initializing. RAM available for dynamic allocation:
I (489) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (495) heap_init: At 3FFBA488 len 00025B78 (150 KiB): DRAM
I (501) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (507) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (514) heap_init: At 40092D6C len 0000D294 (52 KiB): IRAM
I (520) cpu_start: Pro cpu start user code
I (203) cpu_start: Chip Revision: 1
W (203) cpu_start: Chip revision is higher than the one configured in menuconfig. Suggest to upgrade it.
I (207) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
MicroPython v1.12 on 2019-12-20; ESP32 module with ESP32
Type "help()" for more information.
```

I can see the micropython is running somehow. But I can not use it. No key is working except the minicom shortcuts

robert-hh, 2020-04-30:
That looks like a flow control issue. Did you sett flow control off?

nejdetckenobi, 2020-04-30:
Yes I did. No luck. Something is fishy there.
Is there any further information I can provide to you?

robert-hh, 2020-04-30:
You could try a loopback test. If you hold the ESP32 in reset and connect TX and RX (GPIO1 and GPIO3) at the board, do you see the typed characters echoed?

nejdetckenobi, 2020-04-30:
No I don't.

robert-hh, 2020-04-30:
Since the firmware update worked, which required duplex communication, it seems to be a problem with minicom or with the USB device.

nejdetckenobi, 2020-04-30:
Okay, thank you very much. I thought that I eliminated the USB device error option but it is also a possibility. Since this issue isn't related to micropython anymore, I am closing this issue.
I will go buy another one and try to work on it.

closed 2020-04-30 by nejdetckenobi