micropython/micropython #6017: Timing of next release?

jonsmirl, 2020-05-07:
The current release (not master) of micropython is using the commit id of the ESP32 beta 4 SDK. At the time this release was made esp32 support had compile issues. Those issues were quickly fixed in micropython master, however the release tag is in front of those fixes. lv_micropython is only ported to releases of micropython so those build issues got ported into lv_micropython.
https://forum.littlevgl.com/t/esp32-micropython-build-errors-lots/1754

These issues can be fixed with a few edits, but a new micropython release would cure the problem by moving onto the released ESP32 4 SDK.

dpgeorge, 2020-05-08:
We don't have a scheduled release timeline (maybe we need one...) and because everything is done on the master branch it can be hard to coordinate working towards a release and adding new features (maybe that workflow needs to change...).

Regardless, there does need to be a new release sooner rather than later.  And I'd like it to be very soon.

tve, 2020-05-08:
IMHO it would be nice to get the various TLS and non-blocking socket fixes which culminate in adding TLS to uasyncio into MP for the next release..

embeddedt, 2020-05-12:
Is the `master` branch considered stable enough at any point in time to work off of? Perhaps `lv_micropython` should just use that instead.

(cc @amirgon)

dpgeorge, 2020-05-13:
> Is the `master` branch considered stable enough at any point in time to work off of?

The master branch is the development branch, so at any given commit it may have new bugs or regressions.  But the tests are extensive and get very good coverage, and the CI covers a lot, so master is usually quite stable.

jonnor, 2024-07-03:
There has been many releases since 2020! This issue can be closed

closed 2024-07-04 by jonsmirl