micropython/micropython #6021: [Question] ESP32/ESP-IDF know how

montyx99, 2020-05-09:
Hey guys!

I really need a classic Bluetooth function in my ESP32 codebase. I've read in the ESP32 documentation that, the MP is built on-top of ESP-IDF. Is there any way, to write my functionality with ESP-IDF framework, and use it from my MP codebase?
Or this should be much more better, if I could use the arduino bluetooth library of ESP32.
Is there any way to use external binaries or C++ libs?

Thanks a lot

tve, 2020-05-09:
Please use https://forum.micropython.org for questions. The issue tracker is for bugs. Thanks!
(Sorry, I don't have an answer for your question.)

montyx99, 2020-05-09:
> Please use https://forum.micropython.org for questions. The issue tracker is for bugs. Thanks!
> (Sorry, I don't have an answer for your question.)

 I'm so sorry

closed 2020-05-09 by montyx99

zeilenschubser, 2020-05-09:
@montyx99 https://forum.micropython.org/viewtopic.php?t=8309