micropython/micropython #6023: no input for repl with dupterm

uraich, 2020-05-10:
On my esp32 and the esp8266 port REPL does not seem to ask for input except for slot 0.
webrepl shows output but does not take input. Same if I try to use a second serial port. Same for utelnetserver. Is this a dupterm problem?
Working with the newest micropython commit  v1.12-447-gab4e19770-dirty

tve, 2020-05-10:
By default only dupterm slot 0 is configured on the esp32. You have to build your own firmware to increase the number of slots.

uraich, 2020-05-10:
Hi Thorsten
I saw that  MICROPY_PY_OS_DUPTERM was set to (1) which tells me that at least 2 terminals should be allowed? Changing it to a bigger number also does not help.
Or am I looking at the wrong setting?

tve, 2020-05-10:
You'll have to trace through, but when I looked at it I concluded that a value of 1 means there's one slot, and numbering starts at 0. But I could be wrong...

dpgeorge, 2020-05-11:
The value of `MICROPY_PY_OS_DUPTERM` specifies the number of slots.  On esp8266 there are 2, on esp32 there is 1 (which arguably should be increased...).

Stock esp8266 firmware should work fine with WebREPL.  To get further help here you'll need to give all steps to reproduce the problem you see.

uraich, 2020-05-11:
Hi Damien,
So... I tried with ESP8266. Cloned micropython and started from scratch.
In mpy_cross did make
in ports/esp8266 did make
When trying make erase I get an error that /dev/ttyACM0 does not exist: Replaced it by /dev/ttyUSB0. Then erase ok. (in Makefile of ESP32 this is already /dev/ttyUSB0)
Did make deploy and got a warning that --verify is deprecated. Removed it in deploy. I also had to change qio to dio to get a workable micropython for the WeMos D1 8266.
Connected to micropython through /dev/ttyUSB0 thonny (or minicom)
import webrepl_setup
Set to (E) enabled and set password
Reboot
Started WiFi
import webrepl
webrepl.start()
I get:
WebREPL daemon started on ws://192.168.4.1:8266
WebREPL daemon started on ws://192.168.0.31:8266
Started webrepl in normal mode

Started firefox with file ..../webrepl.html
Try to connect to ws://192.168.0.31:8266
Get: 
Welcome to MicroPython!
password
I enter the password and get
WebREPL connected
>>>
When trying to type any command in the WebREPL command window, nothing happens!
This is the same effect on the ESP32
To investigate further I tried on the ESP32 to connect a second serial terminal to UART2 of ESP32.
I increased the number of slots to 3 on the ESP32.
I connect the second serial line to the ESP32 through minicom -D /dev/ttyUSB1
Then I try this:
from machine import UART
import uos
uart2 = UART(2,115200)
uos.dupterm(uart2)
I get the prompt on the second minicom terminal. Anything I type on /dev/ttyUSB0 I see on minicom connected to /dev/ttyUSB1. Typing anything in minicom on ttyUSB1 has no effect.
Hope this helps to trace down the problem. I will also try on my side
Uli

uraich, 2020-05-11:
One more information: I tried micropython-1.2.tar.xz from the download page with the esp8266 and there webrepl works as expected. This means that a recent change, probably to dupterm must have broken the input part of webrepl.

uraich, 2020-05-12:
webrepl is ok now. Maybe the reason why it didn't work was that I used the command window of thonny as the first serial line? Using minicom it works ok on both ESP8266 and ESP32.
The second serial line still doesn't work but I will first check that I get input from the line itself.

strobo5, 2024-11-06:
I think I'm seeing the same issue with a TTGO T-Beam v1.0 (ESP_GENERIC-SPIRAM v1.24.0). The normal REPL is fixed to UART 0, all good. I'm initializing a second UART (either number 1 or 2) where I have a TTL-to-RS232 converter and pass this object to `os.dupterm()`. Now everything I do on UART 0 is duplicated on the second UART, and the REPL seems to slow down to match its slower speed. Promising!

But sending input from the second UART has no effect.

I can do `read()` on the second UART and get all the buffered characters, so the UART seems to be OK. Also the [requirement](https://docs.micropython.org/en/latest/library/os.html#os.dupterm) that `readinto()` returns `None` when no data is available seems to be fulfilled, and I think this UART "stream" is non-blocking..

I also tried building an image with `#define MICROPY_PY_OS_DUPTERM (2)`. I can now use `os.dupterm(uart2, 1)` to put the second UART on another slot than 0. Otherwise it doesn't seem to change anything - same problem.

This is what I do:
```
from machine import UART
uart2 = UART(2, baudrate=19200, tx=14, rx=13)

import os
prev = os.dupterm(uart2)
print(f"Previous dupterm slot 0 stream: {prev}")
```

Any ideas, or maybe a hint where i can start looking?

strobo5, 2024-11-09:
I tested the same on an RP2040 (Raspberry Pi Pico W) with micropython releases v1.23.0 and v1.24.0. It's working there. Whatever I send into the second UART appears on the REPL, as you would expect.

Back on the ESP32 I checked the WebREPL. It works. Only what's coming from UART is not being picked up.

```
>>> uart2.read()
b'help()\rwywhyamIstuckinthisbufferahhhhhh'
```

robert-hh, 2024-11-09:
It seems that dupterm is not told that data is incoming. You can do so using the UART.irq() method by adding the line below after enabling dupterm.

`uart.irq(trigger=uart.IRQ_RXx, handler=os.dupterm_notify)`

That works, but was not needed in earlier times. webrepl.py uses a similar mechanism where the terminal is notified by a socket event.

strobo5, 2024-11-09:
Hey yes this works for me too! :smiley:  I tried both ways with v1.24.0, both work.

robert-hh, 2024-11-09:
IRQ_RXIDLE is meant for situation where one receives data blocks of known maximum length and then stops, like in the PPP protocol or when reading data from a GPS sensor Then the IRQ_RXIDLE interrupt  will be called when the message is complete. One has to set rxbuf to match the worst case message size.