micropython/micropython #6026: ESP32 utime.ticks_us() and utime.ticks_ms() functions return unexpected results

LDmicroGitHub, 2020-05-10:
Used by MicroPython v1.12 on the ESP-WROOM-32 board.  

To test execute:  
`import utime`  
`print(utime.ticks_us(), utime.ticks_ms())`  
62179146 7578371  

But I expected to get something like  
**62179**146  **62179**  
with a difference of 1000 times.

Is this a bug or a special feature?

tve, 2020-05-10:
Sadly your assumption is incorrect. There is a 64-bit microseconds since boot counter, call it `us`. `ticks_us = us & 0x3FFFFFFF` and `ticks_ms = (us / 1000) & 0x3FFFFFFF`. The `0x3FFFFFFF` causes the time to wrap around at 30 bits, which is the max that fits into a "small int" in MP, i.e., that fits into one word without heap allocation.

LDmicroGitHub, 2020-05-10:
1) I understand the matter of the difference:  

In esp-idf\components\esp_timer\include\esp_timer.h 
...  
`int64_t esp_timer_get_time(void);`  
...  
In microPython\ports\esp32\mphalport.c   
...  
`uint32_t mp_hal_ticks_ms(void) {`
`    return esp_timer_get_time() / 1000;`
`}`  

`uint32_t mp_hal_ticks_us(void) {`  
`    return esp_timer_get_time();`  
`}`  
...  
Thus, we get different results when truncating int64 to int32 and int64/1000 to int32.

2) But why is the mask 0x3FFFFFFF and not 0x7FFFFFFF?

`sys.maxsize`
2147483647
`"%x" %(sys.maxsize)`
'7fffffff'

tve, 2020-05-10:
`sys.maxsize` tells you that you're on a 32-bit platform. In MP 31-bit _signed_ integers can be represented in a 32-bit word. Ticks are unsigned and thus there are only 30 bits available. (Integers larger than that get represented as big-ints on the esp32).

LDmicroGitHub, 2020-05-11:
Thanks a lot.

closed 2020-05-11 by LDmicroGitHub