micropython/micropython #6027: Way to check if serial is connected

redraw, 2020-05-11:
Hi, I've seen that CircuitPython provides a way to check if serial USB is connected. There is a [supervisor.runtime.serial_connected](https://circuitpython.readthedocs.io/en/3.x/shared-bindings/supervisor/Runtime.html#runtime-supervisor-runtime-information) method.

In my case, I find it useful to be able to check that because my board needs to be deepsleep 99% the time, just after I turn it on, and I need to have perfect timing to be able to enter raw repl mode and achieve a connection by serial USB from my laptop.

dpgeorge, 2020-05-11:
This question is better suited to https://forum.micropython.org

In short: if the board you use has native USB (eg an stm32-based board) then something like `pyb.USB_VCP().isconnected()` will do the trick.  Otherwise if it's a board with a USB-UART convertor then it's not possible to do.

closed 2020-05-11 by dpgeorge

tve, 2020-05-11:
(I didn't see the OP repost on the forum...)

On an esp32 or esp8266 it _may_ be possible to tell by switching the uart RX to a gpio with pull-down. If reading the pin returns a 1 then something is driving it, specifically the USB-serial chip. If reading returns a 0 then nothing is driving it. This detection relies on the USB chip being powered down or in sleep mode when USB is disconnected. Whether this is possible depends on the USB chip and how it's powered (if it is possible at all).

redraw, 2020-05-11:
Sorry, should have posted on the forum first. I'm using an esp8266, I take the advice @tve